sarxos/webcam-capture #372: Strumieniowanie wideo z Raspberry PI

magocz, 2015-07-28:
Cześć,
Mam plan zrobić serwer na Raspberry Pi, ktory bedzie strumieniował obraz wideo do innej aplikacji napisanej w Java Swing. Masz jakis pomysł jak to najlepiej zrobic, aby móc wybierac dwie opcje: ostry i wysokiej jakosci obraz, ale z opoznieniem lub mniej ostry obraz, ale za to bez opoznienia?

Pozdrawiam!

sarxos, 2015-07-30:
Witam.

Jeśli chcesz to zrobić przy pomocy Webcam Capture API, to uprzedzam, że na RasPi prędkość pobierania obrazu z kamerki absolutnie nie powala (od 2 do 5 ramek/s w rozdzielczości VGA, testy na RasPi model B, na B+ nie testowałem). Ma to związek z małą przepustowością GPIO w połączeniu z USB i Java. Mniejsze rozdzielczości będą miały większą przepustowość, jednak nie wiem jakie parametry potrzebujesz. Musiałbyś wykonać testy jak wygląda prędkość pobierania ramek o mniejszej rozdzielczości (QVGA i QQVGA).

Obejściem może być również wykorzystanie modułu kamery, który daje solidnego kopa do performance (jeśli streamujesz bezpośrednio przez MMAL, niestety proxowanie przez Javę rozwala performance na łopatki), więc jeśli potrzebujesz naprawdę szybki i dobrej jakości obraz, to sugeruję by na RasPi postawić prosty serwer napisany w Javie, która odpali ```raspivid``` [(link)](http://www.raspberry-projects.com/pi/pi-hardware/raspberry-pi-camera/streaming-video-using-vlc-player) czytający obraz z modułu kamery, a po stronie klienta wykorzystać VLCj do wyrenderowania strumienia w panelu. Po stronie klienta możesz wykorzystać zmodyfikowany ```VlcjDevice``` z [tego postu](https://github.com/sarxos/webcam-capture/issues/355), który umożliwia renderowanie RTSP w Webcam Capture API (czyli w ```WebcamPanel```, który dziedziczy z ```JPanel```). Oczywiście to jedynie idea, nie testowałem jeszcze takiego rozwiązania, jednak sprawdzałem, że streaming przez ```nc``` do ```mplayer``` nawet na rozdzielczości HD działa bardzo szybko, więc domyślam się, że w VLC będzie podobnie.

magocz, 2015-07-30:
Wielkie dzieki! W następnym tygodniu będę to próbował to zaimplementować i testować, dam znać jak mi poszło!

sarxos, 2015-08-24:
Ok, pozdrawiam i zamykam ticket :) Powodzenia!

closed 2015-08-24 by sarxos