sarxos/webcam-capture #374: Not running in eclipse?

Bazmundi, 2015-08-04:
What gives?

```
[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:879)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:856)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:834)
	at DetectWebcamExample.main(DetectWebcamExample.java:8)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.util.concurrent.FutureTask.report(Unknown Source)
	at java.util.concurrent.FutureTask.get(Unknown Source)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 5 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NoClassDefFoundError: org/bridj/cpp/CPPObject
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(Unknown Source)
	at java.security.SecureClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.access$100(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Caused by: java.lang.ClassNotFoundException: org.bridj.cpp.CPPObject
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	... 17 more
```

sarxos, 2015-08-04:
You are missing required JARs in classpath.

sarxos, 2015-08-04:
From the project main page:

https://github.com/sarxos/webcam-capture#download

> This ZIP file contains Webcam Capture API itself and all required dependencies (in ```libs``` directory)

The **required dependencies** are JARs you have to put in your application's classpath.

closed 2015-08-04 by sarxos

Bazmundi, 2015-08-04:
Weird. I had added the additional libraries from their source websites, having mistakenly assumed the libs supplied where for building your jar and not running it).  But yes, once I added the jars supplied it appears to run.  I just need to work out how to tell it to use my usb camera over the default windows tells it to use.

So, thank you.

Bazmundi, 2015-08-04:
Yep.  Tried PureDeviceTest to see if that would list my webcam but does not???  I get a AVerMedia 7231 Analog (which is not physically connected) but I don't get my Bloggie??

My actual camera list (via a Processing language script) is:
name=MHS-FS2,size=640x480,fps=5
name=MHS-FS2,size=640x480,fps=30
name=MHS-FS2,size=320x240,fps=5
name=MHS-FS2,size=320x240,fps=30
name=MHS-FS2,size=1280x720,fps=5
name=MHS-FS2,size=1280x720,fps=30
name=AVerMedia H753 Hybrid Analog/DVBT,size=720x576,fps=25
name=AVerMedia H753 Hybrid Analog/DVBT,size=720x480,fps=30

Quirk for me is the AVerMedia does not turn up on my devices page whilst the MHS-FS2 does???

Funny thing this bloggie, Processing video library can see it but won't capture from it, your library doesn't appear to be able to see it?  Should the PureDeviceTest  have listed both the MHS-FS2 and the AVerMedia?