sarxos/webcam-capture #375: doesnt work on raspberry

Kolonka, 2015-08-06:
see here -> http://stackoverflow.com/questions/31844531/library-webcam-capture-by-sarxos-not-working-on-raspberry

sarxos, 2015-08-06:
Hi @Kolonka,

People did experience problems with Raspberry Pi in the past using default driver. The solution in your case may be replacing driver by a different one, e.g. [webcam-capture-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j) which seems to be the most stable on RasPi.

The JARs you have to include in your classpath:

* [v4l4j-0.9.1-r507.jar](http://search.maven.org/remotecontent?filepath=com/github/sarxos/v4l4j/0.9.1-r507/v4l4j-0.9.1-r507.jar)
* [webcam-capture-driver-v4l4j-0.3.11-20150713.101304-10.jar](https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.11-SNAPSHOT/webcam-capture-driver-v4l4j-0.3.11-20150713.101304-10.jar)

The webcam-capture-driver-v4l4j is only for Linux and for now it was compiled for x64 and armhf architectures. In case of x32 or armel you will have to compile it on your own so please verify what arch your RasPi is - either armel or armhf.

The code:

```java
static {
    Webcam.setDriver(new V4l4jDriver()); // set driver to be used
}

public static void main(String[] args) {
    JFrame frame = new JFrame("V4L4J Webcam Capture Driver Demo");
    frame.add(new WebcamPanel(Webcam.getDefault()));
    frame.pack();
    frame.setVisible(true);
    frame.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
}
```

Kolonka, 2015-08-06:
i am using standard raspbian from https://www.raspberrypi.org/downloads/
when i test your code, i get noclassdeffounderror ...webcamdriver

i imported the following libs:

    birdj-0.7.0
    slf4j-api-1.7.2
    v4I4j-0.9.1-r507
    webcam-capture-0.3.10
    webcam-capture-driver-v4I4j-0.3.11-20150713.101304

sarxos, 2015-08-06:
You need these JARs:

* slf4j-api-1.7.2
* v4I4j-0.9.1-r507
* webcam-capture-0.3.10
* webcam-capture-driver-v4I4j-0.3.11-20150713.101304

Please provide detailed exception. The "noclassdeffounderror" or "byclassnotfoundexception" are meaningless without the class names.

sarxos, 2015-08-06:
Just FYI, http://qablog.practitest.com/2008/12/principles-of-good-bug-reporting/

Kolonka, 2015-08-06:
ah.. i run the wrong artifact, its taking pictures now... but there is a fail with your logger:

    SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
    SLF4J: Defaulting to no-operation (NOP) logger implementation
    SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

i solved this last time by adding another slf4j library

sarxos, 2015-08-07:
Indeed, this warning tells you that [SLF4J](http://www.slf4j.org/) has no binding added to the classpath. You can either use [slf4j-simple](http://mvnrepository.com/artifact/org.slf4j/slf4j-simple) or some more sophisticated one like [Logback](http://logback.qos.ch/) which is my favourite one.

I haven't added binding to the classpath on purpose because most people does not need it and I had reports in the past about making logs less verbose (now you can control verbosity on your own through the logger configuration).

I'm glad the issue is resolved. In case of any additional issues please do not hesitate creating new ticket.

closed 2015-08-07 by sarxos