sarxos/webcam-capture #376: Can't see usb web cam?!

Bazmundi, 2015-08-06:
Yep. Tried PureDeviceTest to see if that would list my usb webcam but does not??? I get a AVerMedia 7231 Analog but I don't get my Bloggie??

My actual camera list (via a Processing language script) is:
name=MHS-FS2,size=640x480,fps=5
name=MHS-FS2,size=640x480,fps=30
name=MHS-FS2,size=320x240,fps=5
name=MHS-FS2,size=320x240,fps=30
name=MHS-FS2,size=1280x720,fps=5
name=MHS-FS2,size=1280x720,fps=30
name=AVerMedia H753 Hybrid Analog/DVBT,size=720x576,fps=25
name=AVerMedia H753 Hybrid Analog/DVBT,size=720x480,fps=30

The AVerMedia is graphics and tv tuner card in my Windows 7 machine.  So it is a PCI bus device.

Funny thing this bloggie, Processing video library can see it but won't capture from it, your library doesn't appear to be able to see it? Should the PureDeviceTest have listed both the MHS-FS2 and the AVerMedia?

Bazmundi, 2015-08-06:
Ah very neat!
I tried the WebcamDiscoveryListenerExample and voila!  I get a pint against my MHS-FS2. Even better when TakeSnapshotFromVideoExample runs the MHS-FS2 shows up BUT it is displayed as No Image Available.  So, likely same problem as for Processing library, can "see" camera but cannot capture from it for some reason.

sarxos, 2015-08-07:
Hi @Bazmundi,

The list of available webcams comes from operating system but in case of Windows it seems like some may not be really accessible using classic API. Webcam Capture project should cover most of the UVC-compliant devices. 

Seems like Windows is detecting your MHS-FS2 digital camera as "video device", but this device does not expose UVC-compliant interfaces and therefore neither Processing nor Webcam Capture can access it.

sarxos, 2015-08-07:
In regards to your question:

> I assume I can get to the bitmap somehow?!

You can take picture and write it as bitmap using ```ImageIO```, for example, to save to file:

```java
Webcam webcam = Webcam.getDefault(); // or different
webcam.open();
try {
  BufferedImage image = webcam.getImage();
  if (image != null) {
    ImageIO.write(image, "BMP", new File("sample.bmp"));
  }
} finally {
  webcam.close();
}
```

If you want to access RAW image data, you can use ```webcam.getImageBytes()``` instead.

Bazmundi, 2015-08-08:
Thanks.  The notion that the camera is not UVC compliant is a useful piece
of information.

Bazmundi, 2015-08-08:
In fact I am only interested in the MHS-FS2 and it works with your libraries so thanks.

THe AVerMedia does, however, come up as the default so GetDefault() does not help me.

I got around this by using:
```
List<Webcam> webcams = Webcam.getWebcams(); 
webcam = webcams.get(1);
```
Is there a named method?  I had a poke about but didn't find one on first inspection.

No big deal as the code is working for my against my MHS-FS2.  Doesn't explain why Processing cannot capture against the MHS-FS2.

As an exercise for myself, I will endeavour to pull your library into Processing (mostly for my edification and for fun).

Cheers,
B

sarxos, 2015-08-16:
Hi @Bazmundi, 

In regards to your question:

> Is there a named method? I had a poke about but didn't find one on first inspection.

You may try the following:

```java
public static Webcam getWebcamByName(String name);
```

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/Webcam.java#L1281

closed 2015-08-16 by sarxos

Bazmundi, 2015-08-16:
Yep.  I get told it does not exits though?  I will poke around to see what the problem might be.

Bazmundi, 2015-08-16:
Ah.  All good, sort of.  I needed to swap out to the bleeding edge (0.3.11). However, code "works" (function gets called) but camera name not recognised.  Hmmm.  Will continue to poke around.

MgMed, 2018-02-20:
This worked for me:
`Webcam webcam = Webcam.getWebcams().get(0);`
Juste try another USB port cam not detected.

sarxos, 2018-02-21:
@MgMed,

It should work as long as the operating system is able to detect this camera.

Therefore:

1. Does your operating system detects the camera after you connect it to the other port?
2. Have you close Java application before you switched to new USB port?

MgMed, 2018-02-21:
Yes it was detected on windows but not by the library, after i changed the USB input, the USB CAM was the default one and the integrated CAM become the N°01.
But im my project the USB cam will be the only cam in the system.

@sarxos : do you have a solution or turn over for Disabling the autofocus os set the focus it manually ?

sarxos, 2018-02-21:
In regards to:

> after i changed the USB input, the USB CAM was the default one and the integrated CAM become the N°01

This is unfortunately how it works. This cannot be changed because the order of cameras on the list is controlled by the operating system (applies to Windows, Mac and Linux).

> in my project the USB cam will be the only cam in the my target system

You can also select camera by name:

```java
Webcam.getWebcamByName(String)
```

Or iterate over the cameras to match name pattern:

```java
public static getMyWebcam() {
  for (Webcam webcam : Webcam.getWebcams()) {
    String name = webcam.getName();
    if (/* test if name match your criteria */) {
      return webcam;
    }
  }
  throw new IllegalStateException("Camera not found);
}
```

On application first run you could ask user to select which camera he would like to use and save his choice in some local file. On every subsequent application run you could then read the camera name from this file and access it with the method I described above. This should solve the cameras order problem when camera is switched to different USB port.

> Do you have a solution or turn over for Disabling the autofocus or set the focus it manually ?

No, there is no such possibility by default.

MgMed, 2018-02-21:
Okey Thanks @sarxos , i'll look for an OS command to disabling the autofocus or set it manually.
i'll keep you informed if i find an solution for that.
Regards,