sarxos/webcam-capture #378: problem in webcam integration

rahulala, 2015-08-11:
this type of error shows
Exception in thread "main" java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:97)
	at TakePictureExample.main(TakePictureExample.java:27)
Caused by: java.lang.ClassNotFoundException: org.slf4j.LoggerFactory
	at java.net.URLClassLoader$1.run(URLClassLoader.java:372)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 2 more
Java Result: 1

sarxos, 2015-08-11:
You are missing required dependencies in your classpath. I guess you are not using Maven (I really recommend it, you can avoid such issues in the future), so it's enough to include all JARs from ```libs``` directory that is present in the ZIP file from the project main page (see Download section).

sarxos, 2015-08-11:
Duplicates #158, #185 and many others:

https://github.com/sarxos/webcam-capture/search?q=NoClassDefFoundError%3A+org%2Fslf4j%2FLoggerFactory&type=Issues&utf8=%E2%9C%93

closed 2015-08-11 by sarxos