sarxos/webcam-capture #379: extremely low fps

Kolonka, 2015-08-14:
hi

i am on raspberry with my usb camera and i noticed that it takes about 200 milliseconds to call the method <webcam.getImage()>. <webcam.getFPS()> says 7 fps...

is  my webcam so bad or is it a software problem?

sarxos, 2015-08-14:
Hi @Kolonka, is it Raspberry Pi or normal PC?

sarxos, 2015-08-14:
Hi, sorry. I missed you are on RasPi. My fault. People noticed the problem with RasPi is low throughput from USB to memory buffers - this is the root cause of low FPS.

Kolonka, 2015-08-14:
hi
ok, so the CSI camera will help here?

sarxos, 2015-08-15:
I'm sure RasPi camera module will work much faster with native streaming (e.g. ```raspivid``` and ```netcat``` to ```mplayer```) but I haven't verified the performance in Webcam Capture API which has to copy pixels from memory to ```BufferedImage``` and this operation, at least theoretically, may cause some bottlenecks in the flow.

If you have camera module, feel free to try it, but if you don't have one and need some benchmark before you purchase it, then I can craft some small test on my RasPi B+ to give you the FPS numbers.

I suppose you are interested just in API running with RasPi camera module FPS, not the actual rendering speed on ```WebcamPanel``` (this is quite limited due to no acceleration on Swing component)?

Kolonka, 2015-08-15:
hey

it would be interesting to know how fast the csi camera is. one part of my current project is, to send the live-video from raspberry to a pc via bluetooth (bluecove) and show them there (currently in a jlabel). at the same time the pc is sending sth else to rasp to control it (jxinput). i am using bluetooth 4.0 + EDR Class 1 so it maybe even work with hd-videos.

It's important not to have a big delay (take photo-> send via bluetooth -> show in jlabel) so rasp can be controlled without lags

sarxos, 2015-08-15:
Hi @Kolonka, 

I did the benchmark using [webcam-capture-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j) and [webcam-capture](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture) version from HEAD. The camera I used is [this one](https://www.raspberrypi.org/products/pi-noir-camera/). I hoped to get my hands on [model B+](https://www.raspberrypi.org/products/model-b-plus/) or [model 2 B+](https://www.raspberrypi.org/products/raspberry-pi-2-model-b/), but unfortunately I was able to test only on [model B](https://www.raspberrypi.org/products/model-b/) and this is the one I used.

The FPS strongly depends on the image size. The results are the following:

```
HD720p (1280x720)
FPS series 0: 2.1285653469561514
FPS series 1: 3.119151590767311
FPS series 2: 2.6288117770767614
FPS series 3: 2.6150627615062763
FPS series 4: 2.8137310073157007
FPS series 5: 2.7917364600781687
FPS series 6: 2.6096033402922756
FPS series 7: 3.244646333549643
FPS series 8: 3.236245954692557
FPS series 9: 3.2637075718015667
```

```
VGA (640x480)
FPS series 0: 5.070993914807302
FPS series 1: 6.51890482398957
FPS series 2: 7.788161993769471
FPS series 3: 7.727975270479135
FPS series 4: 7.739938080495356
FPS series 5: 6.067961165048544
FPS series 6: 7.739938080495356
FPS series 7: 7.429420505200595
FPS series 8: 6.289308176100629
FPS series 9: 7.564296520423601
```

```
QVGA (320x240)
FPS series 0: 11.904761904761905
FPS series 1: 22.42152466367713
FPS series 2: 21.459227467811157
FPS series 3: 19.23076923076923
FPS series 4: 21.551724137931036
FPS series 5: 19.37984496124031
FPS series 6: 20.408163265306122
FPS series 7: 21.73913043478261
FPS series 8: 20.242914979757085
FPS series 9: 21.459227467811157
```

The code I used:

```java
import java.awt.Dimension;
import java.io.File;
import java.io.IOException;

import javax.imageio.ImageIO;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

public class CalculateFPSExample {

	static {
		Webcam.setDriver(new V4l4jDriver());
	}

	public static void main(String[] args) throws IOException {

		long t1 = 0;
		long t2 = 0;

		int series = 10;
		int images = 5;

		Dimension resolution = WebcamResolution.QVGA.getSize();
		Webcam webcam = Webcam.getDefault();
		webcam.setCustomViewSizes(new Dimension[] { resolution });
		webcam.setViewSize(resolution);
		webcam.open();

		try {
			for (int k = 0; k < series; k++) {

				webcam.getImage();

				t1 = System.currentTimeMillis();
				for (int i = 0; ++i <= images; webcam.getImage().getType());
				t2 = System.currentTimeMillis();

				System.out.println("FPS series " + k + ": " + (1000d * images / (t2 - t1 + 1)));
			}
		} finally {
			webcam.close();
		}
	}
}
```

I also found some bugs (#382) which I need to look into.

In regards to UDOO, Beaglebone Black, Edison and other alternatives, I haven't confirmed it to be working just out-of-the-box. To make it run the native parts need to be recompiled on these platforms, which may be tricky in some cases.

Kolonka, 2015-08-19:
hi

my camera arrived...
i found this site here: https://www.raspberrypi.org/documentation/raspbian/applications/camera.md
it says, that there should be a lot more fps with the camera module. but ok. currently i am testing the camera module. raspistill -o cam.jpg works. i can see a live view of the camera. but when i the library can't find it.

    import com.github.sarxos.webcam.Webcam;
    import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

    public class CarWebcam {

        static {
            Webcam.setDriver(new V4l4jDriver());
        }

        public static void main(String[] args) {
            Webcam webcam;
            System.out.println(Webcam.getWebcams().size());
            webcam = Webcam.getDefault();
            webcam.open();
        }
    }

it's just returning 0 and nullpointerexception at webcam.open

-> it can't find the webcam module

sarxos, 2015-08-19:
Hi @Kolonka,

It has much higher FPS when using native streaming (not via Java).

Have you had a chance to look at this:

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

Especially:

```plain
sudo modprobe bcm2835-v4l2
```

Kolonka, 2015-08-19:
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
    SLF4J: Defaulting to no-operation (NOP) logger implementation
    SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
    [ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
    [ v4l2-query.c:49 ] Error looking up V4L2 format 34363248. Please submit
    a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
    [ v4l2-query.c:485 ] H264 (0x34363248)
    [ v4l2-query.c:486 ] Please let the author know about this error.
    [ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.

sarxos, 2015-08-20:
Yup, this in fact is the reason why streaming thru Java is much slower than the native one. The native uses h.264 encoding and since Java cannot use it, the next, most sufficient is MJPEG, then YUV, and then RGB, but these consumes much more bandwidth when compared to h.264, thus, decreasing effective FPS.

Kolonka, 2015-08-24:
uhm. do you maybe have a solution for this format issue? it would like to have a working program before rewrite everything in another lang (which will bring up more problems...)

chrbongardt, 2016-04-12:
Is there any workaround for this? :(

sarxos, 2016-04-12:
@chrbongardt, there is no known workaround I'm aware of.

easeway, 2016-10-23:
I have some test results using Rpi2 + Rpi Cam Module. My program is in Go (similar to C++ regarding performance) and using V4L2 API. The interest thing I discovered is, with format H.264, MJPG, the camera can run full speed, and I can easily get 60fps at 640*480. This is also true with RGB/YUYV etc non-compressed format. However, I can only get 4fps with JPEG format, and I observed the current it consumed was higher (>700mA compared to normal 500mA with H.264/MJPG). I shrinked the image size to 320x240, I still got 4fps.