sarxos/webcam-capture #381: Using Eclipse or starting via java -jar the WebCamViewerExample.java blocks when closing via menu

OLFDB, 2015-08-15:
Hi,

I started using the WebCamViewerExample.java in an Eclipse project.
When I export it as runnable jar and start it with java -jar, or I start it in Eclipse and use the Menu to quit the application it blocks.

Here is the thread dump:

2015-08-15 16:55:40
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.31-b07 mixed mode):

"shutdown-hook-1" #29 daemon prio=6 os_prio=31 tid=0x00007ffab4200800 nid=0x19403 waiting on condition [0x0000000140925000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000716f4d408> (a java.util.concurrent.SynchronousQueue$TransferQueue)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.SynchronousQueue$TransferQueue.awaitFulfill(SynchronousQueue.java:762)
	at java.util.concurrent.SynchronousQueue$TransferQueue.transfer(SynchronousQueue.java:695)
	at java.util.concurrent.SynchronousQueue.take(SynchronousQueue.java:924)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:70)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamDisposeTask.dispose(WebcamDisposeTask.java:20)
	at com.github.sarxos.webcam.Webcam.dispose(Webcam.java:426)
	at com.github.sarxos.webcam.WebcamShutdownHook.run(WebcamShutdownHook.java:44)

"notificator-[Microsoft® LifeCam HD-5001 0x14a40000045e076e]" #31 daemon prio=6 os_prio=31 tid=0x00007ffabc26c000 nid=0x1742f waiting on condition [0x0000000138fed000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000717c14018> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2039)
	at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:442)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1067)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1127)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

"frames-refresher-[0x14a40000045e076e]" #28 daemon prio=6 os_prio=31 tid=0x00007ffab5d41000 nid=0x18e5b waiting on condition [0x000000013ca75000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000716f4cfa0> (a java.util.concurrent.SynchronousQueue$TransferQueue)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.SynchronousQueue$TransferQueue.awaitFulfill(SynchronousQueue.java:762)
	at java.util.concurrent.SynchronousQueue$TransferQueue.transfer(SynchronousQueue.java:695)
	at java.util.concurrent.SynchronousQueue.put(SynchronousQueue.java:877)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:68)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice$NextFrameTask.nextFrame(WebcamDefaultDevice.java:60)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.updateFrameBuffer(WebcamDefaultDevice.java:451)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.run(WebcamDefaultDevice.java:480)
	at java.lang.Thread.run(Thread.java:745)

"webcam-lock-[Microsoft® LifeCam HD-5001 0x14a40000045e076e]" #27 daemon prio=6 os_prio=31 tid=0x00007ffab42f8000 nid=0x17603 waiting on condition [0x00000001390f0000]
   java.lang.Thread.State: TIMED_WAITING (sleeping)
	at java.lang.Thread.sleep(Native Method)
	at com.github.sarxos.webcam.WebcamLock$LockUpdater.run(WebcamLock.java:60)

"webcam-discovery-service" #23 daemon prio=6 os_prio=31 tid=0x00007ffab48e6000 nid=0xe40b waiting on condition [0x0000000136a1d000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000716f4cfa0> (a java.util.concurrent.SynchronousQueue$TransferQueue)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.SynchronousQueue$TransferQueue.awaitFulfill(SynchronousQueue.java:762)
	at java.util.concurrent.SynchronousQueue$TransferQueue.transfer(SynchronousQueue.java:695)
	at java.util.concurrent.SynchronousQueue.put(SynchronousQueue.java:877)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:68)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.getDevices(WebcamDefaultDriver.java:80)
[... 130 lines cut ...]
"GC task thread#2 (ParallelGC)" os_prio=31 tid=0x00007ffab4017800 nid=0x2503 runnable 

"GC task thread#3 (ParallelGC)" os_prio=31 tid=0x00007ffab4018000 nid=0x2703 runnable 

"GC task thread#4 (ParallelGC)" os_prio=31 tid=0x00007ffab4018800 nid=0x2903 runnable 

"GC task thread#5 (ParallelGC)" os_prio=31 tid=0x00007ffab4019800 nid=0x2b03 runnable 

"GC task thread#6 (ParallelGC)" os_prio=31 tid=0x00007ffab401a000 nid=0x2d03 runnable 

"GC task thread#7 (ParallelGC)" os_prio=31 tid=0x00007ffab401a800 nid=0x2f03 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007ffab5816000 nid=0x5903 waiting on condition 

JNI global references: 3152

Heap
 PSYoungGen      total 153088K, used 103690K [0x0000000715580000, 0x0000000720000000, 0x00000007c0000000)
  eden space 131584K, 78% used [0x0000000715580000,0x000000071bac2978,0x000000071d600000)
  from space 21504K, 0% used [0x000000071eb00000,0x000000071eb00000,0x0000000720000000)
  to   space 21504K, 0% used [0x000000071d600000,0x000000071d600000,0x000000071eb00000)
 ParOldGen       total 349696K, used 0K [0x00000005c0000000, 0x00000005d5580000, 0x0000000715580000)
  object space 349696K, 0% used [0x00000005c0000000,0x00000005c0000000,0x00000005d5580000)
 Metaspace       used 14470K, capacity 14670K, committed 15104K, reserved 1062912K
  class space    used 1865K, capacity 1931K, committed 2048K, reserved 1048576K

sarxos, 2015-08-15:
Hi, thank you for the report. Can you please give me more details on your operating system version and architecture.

sarxos, 2015-08-24:
I see this thread is blocking the others:

```plain
"atomic-processor-1" #21 daemon prio=6 os_prio=31 tid=0x00007ffab70e0000 nid=0xe607 runnable [0x0000000136b20000]
   java.lang.Thread.State: RUNNABLE
    at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.stopSession(Native Method)
    at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.close(WebcamDefaultDevice.java:391)
    at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.dispose(WebcamDefaultDevice.java:403)
    at com.github.sarxos.webcam.ds.cgt.WebcamDisposeTask.handle(WebcamDisposeTask.java:25)
    at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
    at java.lang.Thread.run(Thread.java:745)
```

This is known issue on Apples. Are you using one?

OLFDB, 2015-08-24:
Yes. Any chance to fix or workaround this?

Regards

Olaf