sarxos/webcam-capture #382: RasPi V4L4J driver does not start for the second time

sarxos, 2015-08-15:
This is in regards to V4L4J driver used on Raspberry Pi with CSI camera module.

At the first time driver works well, but every next run fails with the following stack trace and camera module power turned on.

```plain
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l2-query.c:49 ] Error looking up V4L2 format 34363248. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] H264 (0x34363248)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
	at CalculateFPSExample.main(CalculateFPSExample.java:31)
Caused by: com.github.sarxos.webcam.WebcamException: au.edu.jcu.v4l4j.exceptions.V4L4JException: Error starting the capture
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.open(V4l4jDevice.java:350)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
	at java.lang.Thread.run(Thread.java:724)
Caused by: au.edu.jcu.v4l4j.exceptions.V4L4JException: Error starting the capture
	at au.edu.jcu.v4l4j.AbstractGrabber.start(Native Method)
	at au.edu.jcu.v4l4j.AbstractGrabber.startCapture(AbstractGrabber.java:332)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.open(V4l4jDevice.java:348)
	... 5 more
```

And yet the next one:

```plain
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l2-query.c:49 ] Error looking up V4L2 format 34363248. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] H264 (0x34363248)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
Invalid argument
[ v4l2-input.c:405 ] Unable to set the palette: MJPEG
[ v4l2-input.c:406 ] Please let the author know about this error.
[ v4l2-input.c:407 ] See the ISSUES section in the libvideo README file.
[ v4l2-input.c:606 ] Listing the reported capabilities:
============================================
Querying general capabilities

Driver name: bm2835 mmal
Device name: mmal service -1.-684392064
bus_info: platform:bcm2835-v4l2
version: 3.18.7
Has capture capability
Does NOT have output capability
Has overlay capability
Does NOT have VBI capture capability
Does NOT have VBI output capability
Does NOT have SLICED VBI capture capability
Does NOT have SLICED VBI output capability
Does NOT have RDS capability
Does NOT have tuner capability
Does NOT have audio capability
Does NOT have radio capability
Has read/write capability
Does NOT have asyncIO capability
Has streaming capability

[... 104 lines cut ...]
Control: id: 0x9d0903 - name: Compression Quality - min: 1 -max: 100 - step: 1 - type: 1(Integer) - flags: 0 ()
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
	at CalculateFPSExample.main(CalculateFPSExample.java:31)
Caused by: com.github.sarxos.webcam.WebcamException: au.edu.jcu.v4l4j.exceptions.ImageFormatException: Image format MJPEG not supported
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.open(V4l4jDevice.java:330)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
	at java.lang.Thread.run(Thread.java:724)
Caused by: au.edu.jcu.v4l4j.exceptions.ImageFormatException: Image format MJPEG not supported
	at au.edu.jcu.v4l4j.AbstractGrabber.doInit(Native Method)
	at au.edu.jcu.v4l4j.AbstractGrabber.init(AbstractGrabber.java:189)
	at au.edu.jcu.v4l4j.JPEGFrameGrabber.init(JPEGFrameGrabber.java:105)
	at au.edu.jcu.v4l4j.VideoDevice.getJPEGFrameGrabber(VideoDevice.java:618)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.open(V4l4jDevice.java:328)
	... 5 more
```

mat0pad, 2017-09-08:
Have this been fixed?

mat0pad, 2017-09-08:
I get the same using RPi 3 Model B.
First run goes fine. Then quit and rerun -> error thrown

sarxos, 2017-09-09:
This has not been fixed and I'm not working on this. Can anyone take a look if non-modified v4l4j works well on RasPi? The source code is here:

https://code.google.com/archive/p/v4l4j/source/default/source

And here are some details on how to start with it on RasPi:

https://code.google.com/archive/p/v4l4j/wikis/GettingStartedOnRPi.wiki

haoyun1129, 2017-11-03:
@sarxos 
I have download and built the v4l4j on my Pi but how to replace it in webcam-capture?

sarxos, 2017-11-03:
Here is the fork I used to build v4l4j JAR used by webcam capture driver:

https://github.com/sarxos/v4l4j

After you clone it into your local git repo, please replace these files with the ones you compiled on your RasPi:

https://github.com/sarxos/v4l4j/tree/master/src/main/resources/META-INF/native/linux-armhf

After this is done go to the repository root directory and run ```mvn clean install```. This will create JAR in ```target``` directory. Just put this JAR in your application's classpath and remove the old one beforehand.

That is all.