sarxos/webcam-capture #383: getWebcamByName problem.

Bazmundi, 2015-08-16:
I have imported 0.3.11 library into Processing and entered the following code.
```
import com.github.sarxos.webcam.Webcam;

void setup(){
  
    for (Webcam webcam : Webcam.getWebcams()) { println(webcam.getName()); }
 
    Webcam webcam = Webcam.getWebcamByName("MHS-FS2");
    if (webcam != null) {
      System.out.println("Webcam: " + webcam.getName());
    } else {
      System.out.println("No webcam detected");
    }
  noLoop();
}

void draw() {
}
```

The loop through cams returns my "MHS-FS2".  However, the call getWebcamByName("MHS-FS2") fails with a null.

sarxos, 2015-08-24:
Hi @Bazmundi,

On the OS level, every webcam is internally recognized via ID (1, 2, 3, etc). This is a number that is given by OS and it's used to address specific webcam. The webcam name is something additional, that Webcam Capture API extracts additionally so users sees the webcams by the "name + ID" instead of just "ID" (usability feature to have "Microsoft Webcam 1" instead of just "1"). 

You can find out what are the cameras names by doing:

```java
for (Webcam w : Webcam.getWebcams()) {
  System.out.println(w.getName());
}
```

The above is the "real" name that should be used to find specific webcam via the call to ```getWebcamByName(...)```.

You may ask why we cannot return single instance when calling this method? This is because you may have many webcams of the same model connected to the USB hub.

Bazmundi, 2015-08-25:
Yep.  Gotcha.  I do note that I did get stung, having downgraded from Windoze 7 to Windoze 10, as my cameras are listed in reverse order between the two versions of the OS.  The real quirk being that the Processing Video library returns entirely a different name for the camera as it appears to print a name for each resolution of the camera and no index.  No matter, I am using your library and thankfully yours actually works ;) 

Well done!

sarxos, 2015-10-03:
The Processing is printing supported formats together with camera names. I didn't want to do that because not all capturing frameworks that can be used under the hood of Webcam Capture API support getting info on supported formats. E.g. [OpenIMAJ](http://www.openimaj.org/) does not have it and I'm pretty almost sure that [OpenCV](http://opencv.org/) does not support it as well. The other ones like [GStreamer](http://gstreamer.freedesktop.org/), [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html), etc. has pretty nice API for this, but I wanted to keep Webcam Capture API to be common in regards to framework is being used.

Just FYI, I'm closing this ticket. If you have any further comments in regards to this issue, please do not hesitate to drop them here.

closed 2015-10-03 by sarxos