sarxos/webcam-capture #384: nothing happens after the message webcam open

KumarDhumke, 2015-08-17:
Hi there,
I am using webcam capture api to access an USB industrial webcam. I have installed its driver on my windows 7 pc. 
When I connect the camera and run the example WebcamViewerExample.java, then I get a black window with name of the webcam at the top and FPS:0.0 at the bottom. In the console I get the message webcam open. 
If I stop the execution and runs it again, it works fine. As a workaround to this issue I have done the following - 
                if(webcam.getFPS()==0.0){
                    System.out.println("FPS=0");
                    webcam.close();
                    Thread.sleep(37);
                    webcam.open();
                }

This is working fine but it does take 30 seconds to get the streaming. Could you please help me in resolving this?
Thanks,
Kumar

sarxos, 2015-08-24:
Hi @KumarDhumke,

The Webcam Capture API internals uses various mechanisms to access webcams depending on the capture driver you've choose (don't confuse it with hardware driver). If no other capture driver is provided, the API uses default one which is basically refined [OpenIMAJ](http://www.openimaj.org/) that uses [VideoInput](https://github.com/ofTheo/videoInput) library by Theodore Watson to access Windows API. VideoInput uses DirectShow to create capture graphs that are using either PnP or dedicated webcam hardware drivers (the software you install from the CD you get together with camera).

Since you are using some kind of dedicated hardware driver for this camera, I suppose that issue is somewhere in the layer between VideoInput and the hardware driver, so what I suggest is to replace the framework to use different capturing driver, e.g. the one working on top of [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html):

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-jmf

By doing this you can bypass VideoInput and use Oracle implementation wrapper for DirectShow capture graphs. If the issue persist even after the JMF is used, that may suggest that something is wrong on the connection between the OS and the hardware drivers.

```java
Webcam.setDriver(new JmfDriver());
```

The above is enough to switch from default capturing driver that uses OpenIMAJ to the one using JMF.

KumarDhumke, 2015-08-24:
Thank you Bartosz. We will give it a try and let you know if it works.
Cheers!

sarxos, 2015-10-03:
@KumarDhumke, I'm closing this ticket. Please comment here if you have any further questions in regards to this issue.

closed 2015-10-03 by sarxos