sarxos/webcam-capture #386: Camera compatibilty Question

timmolter, 2015-08-31:
As a video capture newbie, I'm a bit confused about cross platform compatibility. Can you please tell me if I'm understanding this correctly? Based on my research, I should be able to use `webcam-capture` for most UVC compliant devices on either Mac, Linux or Windows. I found a list of UVC cameras [here](http://www.ideasonboard.org/uvc/#devices). Is it really that simple? All those cameras will work on Mac, Linux or Windows using `webcam-capture`, and I can build my own Java app on top?

Specifically I want to capture video from a [dinolite camera](http://www.dinolite.us/products/digital-microscopes/usb) where I develop on MacOSX and deploy on Linux. On Dinolite's [support page](http://www.dinolite.us/support/what-third-party-imaging-software-is-the-dino-lite-compatible-with)
they say:

> Newer Dino-Lite microscopes are also UVC (USB Video Class) compliant and will open video in most applications that support that standard."

So would you think a compatible system would be: dinolite UVC camera + `webcam-capture` + custom Java app on MacOSX and Linux?

Your insight is greatly appreciated!!!

sarxos, 2015-09-08:
Hi @timmolter,

Yes, all UVC cameras should work fine with Webcam capture API. Gross of the UVC devices is properly supported but people reported some issues with some rare devices that usually expect USB 3.0 instead of 2.0 or some very cheap cameras that are only partially standard-compliant.

There are some issues on Mac due to special function of _first thread_ in this operating system and in some cases camera is unable to instantiate when using SWT. On both Linux and Windows everything works fine. However it's very easy to test if your Mac can handle it. Some very simple examples are available here:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture/src/example/java

timmolter, 2015-09-08:
@sarxos Thanks for taking the time to answer! I actually went ahead and bought a Dinolite and a DigiMicro and both work! I was able to use some of your example code to get a live capture working.

closed 2015-09-08 by timmolter