sarxos/webcam-capture #387: Unable to get Webcam-capture to work with CubieBoard 1 A10 ARM SIGSEGV crash

jeronimonunes, 2015-09-02:
```
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver - Video file detected /dev/video0
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Creating V4L4J device from file /dev/video0
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Get video device info
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support BGR conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support JPG conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support RGB conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support YUV conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support YVU conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing JPEG (DISCRETE)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing RGB24 (UNSUPPORTED)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing BGR24 (UNSUPPORTED)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing YUV420 (UNSUPPORTED)
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Best image format match JPEG - 49
[br.ufmg.dcc.luar.follower.HelloWorld.main()] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver V4l4jDriver does not support this feature
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam /dev/video0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Opening V4L4J device /dev/video0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Constructing V4L4J frame grabber
[atomic-processor-1] ERROR com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Resolution mismatch 160x120 vs 320x240, setting new one
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Starting V4L4J frame grabber
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Webcam V4L4J is now open
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb677c3ac, pid=17723, tid=2735006832
#
# JRE version: Java(TM) SE Runtime Environment (8.0_60-b27) (build 1.8.0_60-b27)
# Java VM: Java HotSpot(TM) Client VM (25.60-b23 mixed mode linux-arm )
# Problematic frame:
# V  [libjvm.so+0x3753ac]  LinkResolver::runtime_resolve_virtual_method(CallInfo&, methodHandle, KlassHandle, Handle, KlassHandle, bool, Thread*)+0x74
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/cubie/line-follower/hs_err_pid17723.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
```

Using libaries:
```xml
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture</artifactId>
			<version>0.3.11-SNAPSHOT</version>
			<exclusions>
				<exclusion>
					<artifactId>bridj</artifactId>
					<groupId>com.nativelibs4java</groupId>
				</exclusion>
			</exclusions>
		</dependency>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>v4lvj</artifactId>
			<version>0.9.1-r507</version>
		</dependency>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture-driver-v4l4j</artifactId>
			<version>0.3.11-SNAPSHOT</version>
		</dependency>
```

```java
public class HelloWorld {

	static {
		System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "info");
        System.setProperty("org.slf4j.simpleLogger.log.com.github.sarxos.webcam.ds.v4l4j", "trace");
		Webcam.setDriver(new V4l4jDriver());
	}

	public static void main(String[] args) throws IOException,  InterruptedException {
		Webcam webcam = Webcam.getDefault();
		webcam.open();
		long start = System.currentTimeMillis();
		int i = 0;
		BufferedImage image = null;
		while(System.currentTimeMillis() - start < 2000){
			image = webcam.getImage();
			if(webcam.isImageNew()) i++;
		}
		ImageIO.write(image, "png", new File("hello.png"));
		System.out.println(i/2+" FPS");
		webcam.close();
	}
}

```

sarxos, 2015-09-07:
Hi @jeronimonunes,

Unfortunately I was unable to get my hands on CubieBoard and thus unable to prepare capture driver working on this hardware.

I see you are using V4L4J, but I compiled it on Raspberry Pi.

jeronimonunes, 2015-09-09:
At first I didn't managed to make it work with the default driver, so I ended up trying all drivers. Most of them gave me a Segmentation fault. (I belive FSWebcam was the only one working) But in the end I got back to the default driver and it was working.

As other driver is working, maybe the issue isn't a problem. But if you want to fix the V4L4J driver, I can make one cubie available on internet for you to compile V4L4J. I tried and got the same error.

```xml
	<dependencies>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture</artifactId>
			<version>0.3.10</version>
			<exclusions>
				<exclusion>
					<artifactId>bridj</artifactId>
					<groupId>com.nativelibs4java</groupId>
				</exclusion>
			</exclusions>
		</dependency>
		<dependency>
			<artifactId>bridj</artifactId>
			<groupId>com.nativelibs4java</groupId>
			<version>0.6.3-SNAPSHOT</version>
		</dependency>
		<dependency>
			<groupId>org.slf4j</groupId>
			<artifactId>slf4j-simple</artifactId>
			<version>1.7.12</version>
		</dependency>
	</dependencies>
```