sarxos/webcam-capture #388: example is not working

mcheung63, 2015-09-03:
JFrame frame = new JFrame("JMF Webcam Capture Driver Demo");
frame.add(new WebcamPanel(Webcam.getDefault()));
frame.pack();
frame.setVisible(true);
frame.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.lang.IllegalArgumentException: Webcam argument in WebcamPanel constructor cannot be null!
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:720)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:703)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:693)
	at com.peter.SequenceEncoderDemo.main(SequenceEncoderDemo.java:20)

sarxos, 2015-09-07:
Hi @mcheung63,

Can you give me more details on this? Especially what OS are you using and what arch (32- or 64-bit)?

sarxos, 2015-09-07:
Just FYI, the ```Webcam.getDefault()``` will return ```null``` when webcam cannot be detected. Therefore calling ```new WebcamPanel(null)``` will trigger exception like the one in stacktrace you've provided.

closed 2015-09-12 by sarxos

diogo10, 2016-02-15:
I am running the same code and ge the same error. I would like to test in the default webcam. What's can I do?

**OS X Yosemite**

   ```
 Webcam webcam = null;
        try {
            Webcam.setDriver(new JmfDriver());
            webcam = Webcam.getDefault(10000);

            JFrame frame = new JFrame("JMF Webcam Capture Driver Demo");
            frame.add(new WebcamPanel(webcam));
            frame.pack();
            frame.setVisible(true);
            frame.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);

        } catch (WebcamException ex) {
            Logger.getLogger(FXMLController.class.getName()).log(Level.SEVERE, null, ex);
        } catch (TimeoutException ex) {
            Logger.getLogger(FXMLController.class.getName()).log(Level.SEVERE, null, ex);
        }

```

sarxos, 2016-02-16:
Hi @diogo10,

I see you are using JMF driver. If so, then:

1. Have you properly installed JMF? On Linux and Mac OS this may be very tricky.
2. If so, can you please test this code:

```java
import java.util.Enumeration;
import java.util.Vector;

import javax.media.CaptureDeviceInfo;
import javax.media.CaptureDeviceManager;

public class Test {

	public static void main(String[] args) {
		Vector devices = CaptureDeviceManager.getDeviceList(null);
		Enumeration deviceEnum = devices.elements();
		System.out.println("Device count : " + devices.size());
		while (deviceEnum.hasMoreElements()) {
			CaptureDeviceInfo cdi = (CaptureDeviceInfo) deviceEnum.nextElement();
			System.out.println("Device : " + cdi.getName());
		}
	}
}
```

What the above prints in your case? If the output is like the following:

```plain
Device count : 0
```

That means you have not installed JMF at all, installed it incorrectly, or have it misconfigured.

diogo10, 2016-02-16:
Well, I got 0 like you said. I just added the maven dependency. Well, the user need to install first the drive to so installing our software.....Is it correct? There is a way for delivery everything together?

sarxos, 2016-02-16:
There are different drivers for different capturing frameworks. Some frameworks has to be installed first (like JMF, GStreamer, OpenCV) but some does not have to (e.g. OpenIMAJ, V4L4J). Decision of which framework you would like to use is up to you. By default, if no other driver is selected by the user, Webcam Capture API will use refined OpenIMAJ, however in Mac OS, if you are in Java, only thread with ID = 0 (main) can open/close webcam hardware and therefore people have to make sure to open/close webcam in main - this is known limitation of OpenIMAJ driver.

In regards to JMF - there is no way to bundle it with application without violating their license.

diogo10, 2016-02-16:
Thanks @sarxos  for the explanations.