sarxos/webcam-capture #389: Webcam select

rramoscabral, 2015-09-07:
Hello is possible to select webcam that is intended rather than use the default?

sarxos, 2015-09-07:
Hi @rramoscabral,

First you can list all the webcams you have connected to your PC to learn the names:

```java
for (Webcam w : Webcam.getWebcams()) {
  System.out.println(w.getName());
}
```

The above will output something like (depends on OS, it's different for Win and Linux):

```plain
Integrated Camera /dev/video0
HD Webcam /dev/video1
```

Now when you know what are the camera names you can use them to get desired webcam instance:

```java
Webcam integrated = Webcam.getWebcamByName("Integrated Camera /dev/video0");
Webcam highres = Webcam.getWebcamByName("HD Webcam /dev/video1");
```

Just FYI, when you switch the USB plugs in a different order, the names will change as well.

rramoscabral, 2015-09-07:
Hello, 

Thanks. The problem I was having was that I was using webcam-capture-0.3.10 instead of the last SNAPSHOT.

I tried with 

 
            List<Webcam> webcams = Webcam.getWebcams();
            if (webcams.isEmpty()) {
                System.out.println("No devices found");
            } else {
                
                System.out.println("- - - ");
                webcams.stream().forEach((deviceListVector1) -> {
                    System.out.println(deviceListVector1.getName());
                });
            }
          
            String name = "WebCam SCB-1100N 0";
            for (Webcam webcam : Webcam.getWebcams()) {
                System.out.println(webcam.getName() +  " =/= " + name);
                if (webcam.getName().equals(name)) {
                    System.out.println("OK");
                }
            }

And works.

sarxos, 2015-09-07:
Great. I'm glad it works. This should prompt me to publish 0.3.11 into the Maven Central already...

sarxos, 2015-09-07:
Just FYI, you can simplify your code to:

```java
String name = "WebCam SCB-1100N 0";
Webcam webcam = Webcam.getWebcamByName(name);
if (webcam == null) {
  System.out.println(name + " has not been found"); 
}
```

rramoscabral, 2015-09-07:
Thank you. Today I was to experience the API  I did a project in Netbeans and Swing that allows to select the webcam and take a image. The project is not ring-fenced or Simplified but decided to publish for learning purposes (https://github.com/rramoscabral/webcam-capture-Demo).

I've been experimenting with several webcam and works. But with HP Webcam HD 2300 (A5F64AA) which has a maximum resolution of 720p  but the max resolutions permitted is VGA. 
Is there any way I can try to put the work in maximum resolution?

To be honest I did not see the code of the drivers (webcam-capture-drivers)

sarxos, 2015-09-08:
Hi @rramoscabral,

Yes, you can set HD 720p by:

```java
Dimension[] nonStandardResolutions = new Dimension[] {
  WebcamResolution.HD720.getSize(),
};

Webcam webcam = Webcam.getDefault();
webcam.setCustomViewSizes(nonStandardResolutions);
webcam.setViewSize(WebcamResolution.HD720.getSize());
webcam.open();

BufferedImage image = webcam.getImage();

webcam.close();

System.out.println(image.getWidth() + "x" + image.getHeight());
```

In the same way you can set any other non-standard resolution that your camera supports (unless it's not too big):

```java
Dimension[] nonStandardResolutions = new Dimension[] {
  WebcamResolution.HD720.getSize(),
  new Dimension(123, 789), // weird resolution example
  new Dimension(657, 675), // weird resolution example
};

webcam.setCustomViewSizes(nonStandardResolutions);
webcam.setViewSize(new Dimension(657, 675));

// etc
```

closed 2015-09-08 by sarxos