sarxos/webcam-capture #390: No new image from Camera after the WebcamWebSocketHandler is idle for 3-4 days.

viv816, 2015-09-07:
Hi,
I am using the Websockets API to display image in a web application which captures the image on a button click event and stores it. The system works fine for 2-3 days, 

However after few days "no new image is received from the camera" even though the connection to Websockets is successful when login to the web application. 

It will however work if I restart the WebcamWebSocketHandler server manually.

Need help.

Thanks in advance.

sarxos, 2015-09-07:
Hi @viv816,

Generally, adding some logs should help diagnose what is happening. Can you please add them in ```WebcamCache.java``` in method:

```java
@Override
public void webcamImageObtained(WebcamEvent we) {
	for (WebcamWebSocketHandler handler : handlers) {
		handler.newImage(we.getSource(), we.getImage());
	}
}
```

To test if webcam is still serving new images?

Please take a look into the ```WebcamWebSocketHandler.java```. There are already some logs in the code. You have to enable them. If you don't know how to do that please take a look at:

https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs

viv816, 2015-09-08:
Hi,
I included some debugging statements in code. But I noticed that after a while when there is no connection to websockets, I will get the following stacktrace regarding "connection timed out".

And the application won't work (i.e. no new image received from server) unless I restart the websocket handler.

=========================== Start ==============================
[qtp1427810650-22] INFO WebcamWebSocketHandler - WebSocket closed, status = 1001, reason = null
[webcam-updater-thread-1] ERROR com.github.sarxos.webcam.WebcamExceptionHandler - Exception in thread webcam-updater-thread-1
com.github.sarxos.webcam.WebcamException: Cannot download image
        at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:400)
        at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:318)
        at com.github.sarxos.webcam.ds.cgt.WebcamGetImageTask.handle(WebcamGetImageTask.java:43)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:48)
        at com.github.sarxos.webcam.ds.cgt.WebcamGetImageTask.getImage(WebcamGetImageTask.java:26)
        at com.github.sarxos.webcam.WebcamUpdater.tick(WebcamUpdater.java:229)
        at com.github.sarxos.webcam.WebcamUpdater.run(WebcamUpdater.java:202)
        at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.util.concurrent.FutureTask.run(Unknown Source)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)
Caused by: org.apache.http.conn.HttpHostConnectException: Connection to http://192.168.200.215 refused
        at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:190)
        at org.apache.http.impl.conn.ManagedClientConnectionImpl.open(ManagedClientConnectionImpl.java:294)
        at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:645)
        at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:480)
        at org.apache.http.impl.client.AbstractHttpClient.execute(AbstractHttpClient.java:906)
        at org.apache.http.impl.client.AbstractHttpClient.execute(AbstractHttpClient.java:805)
        at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:374)
        ... 13 more
Caused by: java.net.ConnectException: Connection timed out: connect
        at java.net.DualStackPlainSocketImpl.connect0(Native Method)
        at java.net.DualStackPlainSocketImpl.socketConnect(Unknown Source)
        at java.net.AbstractPlainSocketImpl.doConnect(Unknown Source)
        at java.net.AbstractPlainSocketImpl.connectToAddress(Unknown Source)
        at java.net.AbstractPlainSocketImpl.connect(Unknown Source)
        at java.net.PlainSocketImpl.connect(Unknown Source)
        at java.net.SocksSocketImpl.connect(Unknown Source)
        at java.net.Socket.connect(Unknown Source)
        at org.apache.http.conn.scheme.PlainSocketFactory.connectSocket(PlainSocketFactory.java:127)
        at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:180)
        ... 19 more

 ================================ End =================================== 

Most of the code is default. Delay in WebcamCache is 5000 ms. Am I missing something?

sarxos, 2015-09-08:
Hi @viv816,

Ok, now I see. This is not the WebSocket issue, but the IP camera capture driver problem, namely:

```plain
[webcam-updater-thread-1] ERROR com.github.sarxos.webcam.WebcamExceptionHandler - Exception in thread webcam-updater-thread-1
com.github.sarxos.webcam.WebcamException: Cannot download image
        at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:400)
        at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:318)
        at com.github.sarxos.webcam.ds.cgt.WebcamGetImageTask.handle(WebcamGetImageTask.java:43)
```

When ```IpCamDevice``` fails it probably stops and is unable to start again. I will take a look into this.

closed 2015-09-12 by sarxos

sarxos, 2015-09-12:
Hi @viv816,

I take back what I wrote before. This was neither WebSocket nor capture driver issue. The root cause was the fact that ```WebcamUpdater``` was unable to reschedule when ```WebcamGetImageTask``` threw exception. This is now fixed by https://github.com/sarxos/webcam-capture/commit/0f87892a9b3c4b0dfac579652f6dc4c45ee1a863 and it's already available in a newest SNAPSHOT:

[webcam-capture-0.3.11-20150912.115001-11.jar](https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture/0.3.11-SNAPSHOT/webcam-capture-0.3.11-20150912.115001-11.jar)

Unless you've used some old distribution this should work out-of-the-box after you replace webcam-capture JAR in your application's classpath with the one provided above.

Please let me know if this fixed the issue you've observed.

viv816, 2015-09-14:
Thanks. I will check and let you know.

viv816, 2015-09-22:
Hi,

I monitored the application for 1 week. For first 2-3 days it was working fine but threw the same exception (mentioned in my previous comments) after 2-3 days. However, now there is no problem in getting the new image even after the application is idle for a long time.   

I think the newest SNAPSHOT has fixed it. 

Will monitor more for a longer idle time. 

Thank you very much.

sarxos, 2015-09-22:
Great :) I'm glad to hear that. Take care!

commit 0f87892a (2015-09-12): Fix updater failure when underlying driver throws exception, fixes #390