sarxos/webcam-capture #393: Down sampling a video with WebcamPanel.setFillArea or WebcamPanel.setFitArea

kher52, 2015-09-15:
Hello

I use your API and am very happy with it. Congratulation.

Now I have a Problem with my Camera. The lowest resolution is 1280 x 720. This Pictures are to big for my application. Is it possible to fit this Stream in a smaler frame. I tried setFillArea and setFitArea, but the video has allways the full size.

Thanks for Help in advance!

sarxos, 2015-09-15:
Hi @kher52,

That's weird. Using ```setFitArea(true)``` should cause it to fit into panel's bounds. Maybe your bounds are not limited? In this case panel will automatically have the max possible size.

Example:

```java
import java.awt.Dimension;
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;

public class Test {

	public static void main(String[] args) {

		Webcam webcam = Webcam.getDefault();
		webcam.setCustomViewSizes(new Dimension[] { new Dimension(1280, 720) });
		webcam.setViewSize(new Dimension(1280, 720));

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setFitArea(true);

		JFrame window = new JFrame("Test fit into panel's area");
		window.add(panel);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

The above will display ```JFrame``` that's very large, but when you shrink it down, the image will fit into the panel's size.

[image]

Same with some other size configuration:

[image]

If I add this line:

```java
panel.setPreferredSize(new Dimension(400, 300));
```

The ```JFrame``` that appears will be packed to the smallest possible size, which in this case is the preferred panel size (400 x 300 px). I suppose this is what you need.

```java
import java.awt.Dimension;
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;

public class Test {

	public static void main(String[] args) {

		Webcam webcam = Webcam.getDefault();
		webcam.setCustomViewSizes(new Dimension[] { new Dimension(1280, 720) });
		webcam.setViewSize(new Dimension(1280, 720));

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setFitArea(true);
		panel.setPreferredSize(new Dimension(400, 300));

		JFrame window = new JFrame("Test fit into panel's area");
		window.add(panel);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

Just after I run it, no manual frame resize:

[image]

kher52, 2015-09-16:
Hi @sarxos 
Thank you for your help. My App works now perfect and I get the video in that size I need.

closed 2015-09-16 by kher52

sarxos, 2015-09-16:
Great! Take care :)