sarxos/webcam-capture #394: noclassdeffounderror and classnotfoundexception

aendep, 2015-09-18:
Hi guys
when I am running everything in netbeans it works fine but as soon as i compile everyting and run the jar file it is not working anymore.
when i run the runnable jar from cmd line this error comes up

[image]

Can u guys help me?

Thanks for your efforts

sarxos, 2015-09-18:
Hi @aendep,

You've included ```TakeSnapshotFromVideoExample``` class in package named ```logic``` in NetBeans, but you did not include it in the runnable JAR you are executing.

aendep, 2015-09-18:
How do i include it in the runnable jar?
i included all jars from that class already

sarxos, 2015-09-19:
Hi @aendep,

You can easily find out if the class you are looking is there. Just use WinZIP / WinRAR to open the JAR you are running, go to ```logic``` folder in JAR and check if file ```TakeSnapshotFromVideoExample.class``` is inside this folder. I bet it's not there or either the class or folder has different name.

There are various ways to do include files in the JAR. For beginners the best option is to use some kind of _export_ option from your favourite IDE (Eclipse, NetBeans, IntelliJ, etc):

https://www.youtube.com/watch?v=mE3rbtKm-pk (Eclipse)
https://www.youtube.com/watch?v=1l1KExPKjWE (NetBeans)
https://www.jetbrains.com/idea/help/packaging-a-module-into-a-jar-file.html

If you are the ultimate hardcore you can also use command line (but with many classes this will be pain in the ass):

https://www.youtube.com/watch?v=uOCtUxrhKds

You can also build your project with Maven (all modern IDEs has support for this option) with ```maven-assembly-plugin``` or ```maven-shade-plugin```, but this case if for rather experienced Java developers.

https://www.google.com/search?q=maven+create+runnable+jar

aendep, 2015-09-19:
Hi i checked it and everything was there. only the first letter of the folder was written like this Logic and not logic.
my way to include to jar was already the same way like netbeans. have you got another idea?

sarxos, 2015-09-19:
The package name is very important so ```Logic``` is not the same as ```logic```. Make sure to change this and it should run.

Just FYI, don't forget to add JARs required by Webcam Capture to the classpath. In the screenshoot I do not see them listed.

```
java -cp <required jars goes here> -jar yourcode.jar
```

E.g. when you put all required JARs in ```lib``` directory from where you run ```java``` you can do:

```
java -cp lib/* -jar yourcode.jar
```

JARs required by Webcam Capture API are the same you are using in NetBeans.

sarxos, 2015-09-21:
I'm closing this ticket. Take care!

closed 2015-09-21 by sarxos