sarxos/webcam-capture #395: Support GStreamer 1.x

neilcsmith-net, 2015-09-23:
A notification by way of a feature request :smile:

Over the last couple of weeks I've been trying to fix up GStreamer-Java support for GStreamer 1.x.  Webcam pipelines do seem to be working OK, but it would be good to get some other projects testing.

https://github.com/gstreamer-java/gst1-java-core

If you're interested in providing an updated GStreamer driver I'm happy to help with info.  The API has changed a bit and documentation is a bit sparse at the moment.  There's also a simple webcam example in https://github.com/gstreamer-java/gst1-java-examples (in fact, only example so far)

sarxos, 2015-09-27:
Hi @neilcsmith-net,

This is wonderful news :) I'm really glad for letting me know about [gstreamer-java](https://github.com/gstreamer-java) and will start working on the updated GStreamer Webcam Capture driver as soon as possible.

I was observing another GStreamer Java implementation project so far, but since there was no update for more a year now I didn't do anything to get it working within Webcam Capture API. Here is the link:

https://github.com/octachoron/gstreamer-1.x-java

neilcsmith-net, 2015-09-27:
Hi @sarxos 

Thanks.  Yes, there is also https://github.com/armouroflight/gstreamer1.x-java which I forked back for gst1-java-core.  When Google Code shut down a few of us who were maintainers on the old repository set up the GitHub organisation to try and bring the various efforts together.  I have to admit my own involvement around 1.x support has been minimal until now, but I suddenly found I need it for another project! :smile:  (see latest tweet on https://twitter.com/PraxisLIVE)