sarxos/webcam-capture #397: Unsupported major.minor version 52.0

biswajitroy04, 2015-09-27:
Hi Sir/Madam,
                          My application uses com.github.sarxos.webcam.Webcam libraries and that's why it needs JRE 1.8 and my application is working in my local since I have JRE 1.8 in my local but while I am deploying my application in "http://security-ecam.rhcloud.com/" OpenShift  I am getting below Exception

java.lang.UnsupportedClassVersionError: com/TestJava : Unsupported major.minor version 52.0 (unable to load class com.TestJava)
	org.apache.catalina.loader.WebappClassLoader.findClassInternal(WebappClassLoader.java:2961)
	org.apache.catalina.loader.WebappClassLoader.findClass(WebappClassLoader.java:1210)
	org.apache.catalina.loader.WebappClassLoader.loadClass(WebappClassLoader.java:1690)
	org.apache.catalina.loader.WebappClassLoader.loadClass(WebappClassLoader.java:1571)

So how to deploy and run the app in server using JRE <1.8 

Thanks 
Biswajit Roy

biswajitroy04, 2015-09-28:
I have fixed this by compiling the libraries in JDK 1.7 now this is working in my local tomcat with JVM 1.7 but I am getting below error while deploying the application in openshift.redhat.com. I don't know RedHat environment details :(

which basically says 

Caused by: java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/var/lib/openshift/56081b1789f5cf12fc000062/jbossews/tmp/BridJExtractedLibraries9012739367972665359/OpenIMAJGrabber.so'

StackTrace: 
com.github.sarxos.myWebcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.myWebcam.WebcamException: Cannot execute task at com.github.sarxos.myWebcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124) at com.github.sarxos.myWebcam.Webcam.getWebcams(Webcam.java:816) at com.github.sarxos.myWebcam.Webcam.getDefault(Webcam.java:879) at com.github.sarxos.myWebcam.Webcam.getDefault(Webcam.java:856) at com.github.sarxos.myWebcam.Webcam.getDefault(Webcam.java:834) at com.TestJava.doGet(TestJava.java:38) at javax.servlet.http.HttpServlet.service(HttpServlet.java:620) at javax.servlet.http.HttpServlet.service(HttpServlet.java:727) at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:303) at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:208) at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:52) at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:241) at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:208) at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:220) at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:122) at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:501) at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:171) at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:102) at org.apache.catalina.valves.RemoteIpValve.invoke(RemoteIpValve.java:683) at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:116) at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:408) at org.apache.coyote.http11.AbstractHttp11Processor.process(AbstractHttp11Processor.java:1042) at org.apache.coyote.AbstractProtocol$AbstractConnectionHandler.process(AbstractProtocol.java:607) at org.apache.tomcat.util.net.JIoEndpoint$SocketProcessor.run(JIoEndpoint.java:316) at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145) at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615) at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61) at java.lang.Thread.run(Thread.java:745) Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.myWebcam.WebcamException: Cannot execute task at java.util.concurrent.FutureTask.report(FutureTask.java:122) at java.util.concurrent.FutureTask.get(FutureTask.java:188) at com.github.sarxos.myWebcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116) ... 27 more Caused by: com.github.sarxos.myWebcam.WebcamException: Cannot execute task at com.github.sarxos.myWebcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72) at com.github.sarxos.myWebcam.WebcamProcessor.process(WebcamProcessor.java:140) at com.github.sarxos.myWebcam.WebcamTask.process(WebcamTask.java:46) at com.github.sarxos.myWebcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46) at com.github.sarxos.myWebcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118) at com.github.sarxos.myWebcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36) at com.github.sarxos.myWebcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:1) at java.util.concurrent.FutureTask.run(FutureTask.java:262) at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145) at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615) ... 1 more Caused by: java.lang.RuntimeException: Failed to allocate new instance of type class com.github.sarxos.myWebcam.ds.buildin.natives.OpenIMAJGrabber at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:674) at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:828) at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:755) at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:209) at org.bridj.BridJ.initialize(BridJ.java:970) at org.bridj.NativeObject.(NativeObject.java:23) at org.bridj.StructObject.(StructObject.java:14) at org.bridj.cpp.CPPObject.(CPPObject.java:27) at com.github.sarxos.myWebcam.ds.buildin.natives.OpenIMAJGrabber.(OpenIMAJGrabber.java:64) at com.github.sarxos.myWebcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56) at com.github.sarxos.myWebcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ... 3 more Caused by: java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/var/lib/openshift/56081b1789f5cf12fc000062/jbossews/tmp/BridJExtractedLibraries9012739367972665359/OpenIMAJGrabber.so' at org.bridj.BridJ.getNativeLibrary(BridJ.java:922) at org.bridj.BridJ.getNativeLibrary(BridJ.java:901) at org.bridj.BridJ.getNativeLibrary(BridJ.java:513) at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:627) ... 13 more Mail Sent

sarxos, 2015-09-28:
Hi @biswajitroy04,

Thank you for the report.

I checked the [0.3.10](http://search.maven.org/remotecontent?filepath=com/github/sarxos/webcam-capture/0.3.10/webcam-capture-0.3.10.jar) version available from Maven:

```plain
$ javap -verbose Webcam.class | grep major
$ javap -verbose Webcam.class | grep minor
```

The output is:

```plain
major version: 50
minor version: 0
```

The above points out it's 1.6. I did the same with the [0.3.11-SNAPSHOT](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.11-SNAPSHOT) and effect was the same. I suppose that problematic class in your case was simply ```TestJava``` which has been compiled with 1.8. Just guess this is your project's main class.

In regards to the second issue - this may be caused either by:

1. The fact that you are running this application in cloud env (trying to open UVC webcam on the server side), and the cloud does not give you permission to load native code (OpenIMAJGrabber.so), or
2. The fact that you renamed package to ```myWebcam``` (this should go in pair with the change in corresponding folder in ```src/main/resources``` (I remember BridJ having some problems to locate *.so objects in such a case).

But are you **really** trying to open UVC in a cloud? Or did you want to use some IP camera and simply forget to set appropriate capture driver?

biswajitroy04, 2015-10-01:
Thanks a lot for your reply. I will check all these aspects and yes I need
to modify the codes to use it as an IP camera.

Thanks
Biswajit

sarxos, 2015-10-01:
Hi @biswajitroy04,

Ok, in regards of the major/minor version issue I'm almost 100% sure that ```TestJava``` is to blame.

To switch from UVC to UIP camera just replace the capture driver:

```java
static {
	Webcam.setDriver(new IpCamDriver(new IpCamStorage("cameras.xml")));
}
```
_(keep in mind that you have to have ```cameras.xml``` pre-created, example can be found [here](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/resources/cameras.xml))_

By doing the above the issue with ```OpenIMAJGrabber``` will disappear. FYI, here are some examples using IP cameras:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

And here is small low latency streaming example using IP cameras on a web service backend which transmit images to the GUI through WebSocket.

Take care!

sarxos, 2015-10-01:
I forget to put a link to the example. Here it is:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets

sarxos, 2015-10-03:
@biswajitroy04, I'm closing this ticket. Please comment here if you have any further questions in regards to this.

closed 2015-10-03 by sarxos