sarxos/webcam-capture #398: Live Stream

magocz, 2015-09-29:
Czesc,
Widzac jaka wiedze posiadasz, chcialbym zapytac, jakbys rozwiazal taki problem: chce zrobic monitorowanie budynku za pomoca raspberry. Do niego bedzie podlaczona kamerka internetowa i bedzie postawiony na nim serwer, np. Jboss czy tomcat z oppalona aplikajca webowa, ktora by streamowala obraz z kamerki podlaczonej do PI, do zalogowanych klijentow. Masz jakis pomysl?
Pozdrawiam!

sarxos, 2015-10-01:
@magocz, jesli planujesz rozwiązać to za pomocą RasPi, to musisz mieć świadomość istnienia pewnych ograniczeń:

1. Połączenie Java + RasPi do streamingu obrazu daje wydajności mocno w kość więc efektywnie uzyskasz może streaming rzędu 1-5 FPS z kamerki podłączonej do USB (w zależności od rozdzielczości - im większa tym mniejszy FPS). Jeśli to nie ma być kamera USB tylko kamera IP, to może nie być to już takim problemem, bo IP mają już same z siebie dosyć niskie latency.
2. Absolutnie unikaj ciężkich kontenerów (np. Tomcat, a tym bardziej JBoss). Polecam osadzenie w swojej aplikacji Jetty - robiłem tak nie raz i takie rozwiązanie jest naprawdę lekkie, a otrzymujesz w efekcie kompletny servlet stack.
3. Webcam Capture API na niektórych RasPi działa trochę niestabilnie (nie wiem czemu, próbowałem zreprodukować te problemy u siebie, ale bez efektu). Aby to obejść można zmienić capture driver na bardziej stabilny i powinno być git, jednak wszystkie urządzenia oparte na ARM to dla mojego projektu na razie pole doświadczalne.

Generalnie, wybór lub dyskryminacja tego rozwiązania totalnie zależy od tego jak szybki streaming chcesz uzyskać. Jeśli planujesz rozwiązanie low latency, to RasPi + Java + Webcam capture API mogą spokojnie być użyte by to zaimplementować, jeśli jednak planujesz streaming z prędkością większą niż 1-2 ramki na sekundę to polecam dedykowaną kamerę IP (koszt taki sam jak zakup nowego RasPi w sumie z czasem, który poświęcisz na implementację).

magocz, 2015-10-01:
@sarxos, dobra to wykorzystam Jetty. A co do stremingu, to 5 klatek na sekunde mi wystarczy, bo ma to byc podgladowy stream.

sarxos, 2015-10-01:
@magocz, ku pamięci - 5 klatek to jest absolutne maksimum i to w dodatku w rozdzielczości QVGA (320x240). Dla HD 720p FPS wynosi od 0.2 do 0.5 (jedna klatka na pięć sekund).

sarxos, 2015-10-03:
Hej @magocz, zamykam to zgłoszenie. Powodzenia w implementacji!

closed 2015-10-03 by sarxos

magocz, 2015-10-05:
A i chcialem jeszcze zapytac, czy moge kozystac z Twojego kodu i bibliotek, przy pisaniu mojej pracy inzynierskiej?

sarxos, 2015-10-05:
Oczywiście że tak. Kod jest na licencji MIT, więc możesz z nim zrobić praktycznie co chcesz. Niektóre frameworki, na których zbudowane są capture drivers mogą posiadac inną licencję (np. GPL, BSD, etc), jednak żadna nie jest shareware i nie blokuje wykorzystania w taki czy inny sposób.

magocz, 2015-10-05:
Oki, cieszy mnie to bardzo! ;) Sory, ze tak spamuje tutaj, ale już od trzech dniu próbuje zrobic ten live stream, ale jakoś nie chce mi to zadziałać. Mógłbyś podać, jakiś prosty przykładowy kod, który na podstawie obiektu Webcam, bedzie stremował np. na tomcacie na jakis adres URL.?

sarxos, 2015-10-05:
Generalnie Webcam Capture API nie udostepnia metod stricte do remote streamingu, ktory supportuje jedynie w podsawowym zakresie. To z uwagi na fakt, ze frameworkow do streamingu jest na peczki (szczegolnie polecam [netty](http://netty.io/)).

Przyklady, ktore przygotowalem w ramach projektu:

#### Simple MJPEG server

Uruchamiasz kod, w przeglądarce wchodzisz na http://localhost:8080 i dostajesz obraz z kamerki (oczywiście zamiast localhost używasz IP gdy uruchamiasz to na serwerze):

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/MjpegStreamingExample.java

#### Images Streaming via WebSocket

Bardzo prosty serwer oparty o Jetty, ktory przez WebSocket udostepnia obraz z kilku kamerek (w tym przykladzie sa to kamerki IP, ale mozna uzyc tez embedded lub UVC).

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets

Głowna klasa:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-websockets/src/main/java/WebcamWebSocketsExample.java

HTML:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets/src/main/html

magocz, 2015-10-05:
O super, ale rzuca mi takim wyjatkiem:
java.net.SocketException: Software caused connection abort: socket write error
	at java.net.SocketOutputStream.socketWrite0(Native Method)
	at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:109)
	at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
	at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
	at com.github.sarxos.webcam.WebcamStreamer$Connection.run(WebcamStreamer.java:136)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[streamer-thread-1] ERROR com.github.sarxos.webcam.WebcamStreamer - Error
java.net.SocketException: Socket closed
	at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:116)
	at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
	at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
	at java.io.FilterOutputStream.close(FilterOutputStream.java:158)
	at com.github.sarxos.webcam.WebcamStreamer$Connection.run(WebcamStreamer.java:140)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[streamer-thread-1] ERROR com.github.sarxos.webcam.WebcamStreamer - Cannot close socket
java.net.SocketException: Socket closed
	at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:116)
	at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
	at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
	at java.io.FilterOutputStream.close(FilterOutputStream.java:158)
	at com.github.sarxos.webcam.WebcamStreamer$Connection.run(WebcamStreamer.java:174)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)