sarxos/webcam-capture #399: saving image as base64 encode

swap17, 2015-10-01:
I am using the below mentioned code in applet for saving image in base64 encode string .On creating image from base64 encoded string the image is cropped its not showing the complete image.please suggest.

ByteArrayOutputStream os = new ByteArrayOutputStream();
final BufferedImage bufferedImage = webcam.getImage();
OutputStream b64 = new Base64.OutputStream(os);
ImageIO.write(bufferedImage, "png", b64);
String imageStr = os.toString("UTF-8");

sarxos, 2015-10-01:
Hi @swap17,

Can you please elaborate more on how the image is cropped? Is the vertical or horizontal size smaller?

Moreover, can you check what is the image size you are getting from Webcam:

```java
BufferedImage bufferedImage = webcam.getImage();
System.out.println("width " + bufferedImage.getWidth());
System.out.println("width " + bufferedImage.getHeight());
```

I also suggest to use JPG instead of PNG (the fist one has a lot smaller size after DCT). Check this out:

```java
import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.ByteArrayOutputStream;
import java.io.IOException;

import javax.imageio.ImageIO;

import com.github.sarxos.webcam.Webcam;

public class Test {

	public static void main(String[] args) throws IOException {

		Webcam webcam = Webcam.getDefault();
		webcam.setCustomViewSizes(new Dimension[] { new Dimension(1280, 720) });
		webcam.setViewSize(new Dimension(1280, 720));
		webcam.open();

		ByteArrayOutputStream baos1 = new ByteArrayOutputStream();
		ByteArrayOutputStream baos2 = new ByteArrayOutputStream();

		BufferedImage bi = webcam.getImage();

		ImageIO.write(bi, "JPG", baos1);
		ImageIO.write(bi, "PNG", baos2);

		System.out.println("JPG = " + baos1.size());
		System.out.println("PNG = " + baos2.size());
		System.out.println("ratio = " + (baos2.size() / baos1.size()));
	}
}

```

The output for HD 720p is:

```plain
JPG = 56035
PNG = 1267544
ratio = 22
```

The JPG compression is **22 times more efficient**  than PNG.

sarxos, 2015-10-03:
I'm closing this ticket. Please comment here if you think it should be reopen.

closed 2015-10-03 by sarxos

swap17, 2015-10-03:
Thanks sarox.I had used JPG and it solved the issue

sarxos, 2015-10-03:
Great :) Take care!