sarxos/webcam-capture #400: Support for applet in chrome

swap17, 2015-10-08:
I am facing issue for implementing applet based photo capture in chrome as chrome does not support NPAPI.

Document from chrome suggest as below
"Why NPAPI plugins don’t work now
In the past, many plugins were developed using an old system called NPAPI. Today fewer sites are using NPAPI plugins and they have often caused security risks on websites.

To make browsing with Chrome safer, faster, and more stable, we stopped allowing NPAPI plugins on September 1, 2015.

Plugins that use NPAPI, including Silverlight, Java, and Unity, won’t work. If you want to use a website that uses an NPAPI plugin, you’ll need to use a different web browser."

please suggest how to use applet based photo capture in chrome.

sarxos, 2015-10-08:
Hi @swap17,

Tu cut the story short - there is no possibility to run applets in Chrome. Both Google and Oracle recommend using different browser to run them:

https://java.com/en/download/faq/chrome.xml

To w/a this issue you can try using IE Tab plugin to run applets inside (but this is actually running applet in IE that runs in Chrome):

http://techhelpkb.com/how-to-enable-java-in-google-chrome-45/

However, this is manual operation which require end user actions and in reality it completely breaks applets portability idea.

The other solution is to not using applets at all and implement your browser solution in JavaScript and this last option is the one I would recommend:

**[https://github.com/jhuckaby/webcamjs](https://github.com/jhuckaby/webcamjs)**

Please keep in mind that Oracle restricted use of Applets by requiring CA signed certificates to be bought to run applets on Java 7u51 and above:

https://www.digicert.com/code-signing/java-7-update-51-code-signatures-required.htm

This is pretty big limitation for 99% of applets which were usually self-signed.

closed 2015-10-08 by sarxos

swap17, 2015-10-08:
sarxos , thanks for your suggestion.