sarxos/webcam-capture #401: PS3 Eye not working

gregkotsaftis, 2015-10-13:
I only have a PS3 Eye in my Windows 7 PC.
Althought the cam is working, a simple test reports:
```
13/10/2015 21:39:27,189  INFO com.github.sarxos.webcam.Webcam.getDriver():971 - WebcamDefaultDriver capture driver will be used
13/10/2015 21:39:27,326  WARN com.github.sarxos.webcam.Webcam.getDefault():887 - No webcam has been detected!
```
The code used is:
```
try
{
    Webcam webcam = Webcam.getDefault();
    if( webcam!=null )
    {
        if( webcam.open() )
        {
            ImageIO.write(webcam.getImage(), "PNG", new File("webcam-hello-world.png"));
        }
        else
        {
            System.err.println("---Webcam not opened!");
        }
    }
    else
    {
        System.err.println("---No webcam found!");
    }
}
catch(Exception ex)
{
    ex.printStackTrace();
}
```
My classpath includes `webcam-capture-0.3.10.jar` and `bridj-0.6.2.jar`, is there something wrong with the code or PS3 Eye is not supported?

sarxos, 2015-10-14:
Hi @gregkotsaftis,

I suppose that PS3 Eye camera is not UVC-compliant. I googled a little bit and haven't found any confirmation for it to be UVC-compliant but several posts allows me to conclude that it really isn't. For example it [does not work](http://ubuntuforums.org/showthread.php?t=1555114) on Ubuntu and you need to have dedicated driver for it. That means it does not work on the UVC drivers which are build-in. Same for Apple's Mac - you need special driver to run it. Seems like Windows [need](http://nuigroup.com/forums/viewthread/2921/) dedicated driver as well (maybe not in recent days - on modern systems it may be build in).

gregkotsaftis, 2015-10-14:
I have bought [Sony CL-Eye driver] (https://codelaboratories.com/products/eye/driver/), and the cam works in Windows. I do not know if the driver is a UVC-compliant or not. Still I tried the same example in an **Acer laptop** and I got this exception:
```
HD WebCam 0
com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:330)
	at gr.grekos.dental.main.U.<init>(Unknown Source)
	at gr.grekos.dental.main.MainDesktop.aK(Unknown Source)
	at gr.grekos.dental.main.eJ.actionPerformed(Unknown Source)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6535)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

phrack, 2015-11-01:
My project has had no problems using the PS3eye on Linux (Ubuntu and Arch), but it doesn't work on Windows unless the application is launched from Java Web Start. We have no idea why the latter case works because it seems like it shouldn't work under any circumstance.

On another note, we don't really like the PS3eye for computer vision applications because it's very sensitive to light conditions and gets more "noise" than your average webcam.

sarxos, 2015-11-02:
Hi @gregkotsaftis, @phrack,

Computer vision on Linux machines is pretty straightforward - there is a libv4l2 middleware that resolves compatibility and connectivity issues for us, and all you need to do is to take the v4l api and use it. On Windows there are much more tricks & caveats (e.g. no locking, sharing capture graph between processes), but the fact that PS3eye is working in Java Web Start and not in normal Java application is completely surprising for me!

Is this the one you are talking about?

http://allegro.pl/kamera-sony-eye-tv-ps3-move-nowa-sklep-tanio-i5749465007.html

Can you please confirm, so I can buy this camera and test it by myself (unfortunately I cannot check on Mac, I do not own one)? Or did you switch to other webcam that is working well?

phrack, 2015-11-02:
Yea, for reference here it is on Amazon:

http://www.amazon.com/PlayStation-Eye-3/dp/B000VTQ3LU/ref=sr_1_1?ie=UTF8&qid=1446468927&sr=8-1&keywords=ps3eye

Everything works better on Linux. We tend to have to fight Windows a little and Mac a lot.

I primarily just use the MS Lifecam HD-3000 because it works well on Linux, Mac, and Windows. The PS3eye is a pain even when it does work because it's so sensitive to lighting conditions.

sarxos, 2015-11-02:
Ok, will purchase it and perform some tests in regards to API working in Web Start and not in classic app. This may be interesting. Any other PS3Eye issues are most likely caused by lack of the UVC support.

sarxos, 2015-11-17:
Ok, I have PS3 Eye. It works really nice on my Ubuntu. Now I need to test it on Windows. Will keep you informed.

salocinx, 2015-12-02:
Hi sarxos. Any news regarding ps eye on windows? I really need to use the PS Eye 3 for my project and I would love to use your webcam lib too :-)