sarxos/webcam-capture #403: Xuggler support

mjoycemilburn, 2015-10-22:
I have been trying to get the "Capture video" example to run but have so far failed due to a series of problems with Xuggle.

I'm pretty new to this stuff so the problems may be of my own creation. An Ant build ran but failed with a "failed to encode video" problem. I wasn't confident that I'd got the library versions lined up properly so switched to Maven. This ran into problems even sooner as Netbeans couldn't build a Maven repository for Xuggle. According to a Netbeans forum response the repository URL that the Xuggle doc supplies (http://xuggle.googlecode.com/svn/trunk/repo/share/java/) has no index.

From what I can gather, Xuggle development has stopped. So I'm stuck unless I get a helpful response from the Xuggle forum.

Any suggestions? I'm trying top run this stuff under 64-bit WIndows 10.

Regards, MartinJ

sarxos, 2015-10-23:
Hi @mjoycemilburn,

You are using correct repo URL. Maven repository may have, but does not need to have index. The index lists content available in a repository group, but it's not required because Maven always tries to load artifact defined in dependency and there is no need for it to index whole repo.

I'm not very familiar with NetBeans (I'm using Eclipse), but you should be able to create new Maven project (or import existing one) without having to build any repository.

This is the POM I'm using in examples. It works well from both command line and Eclipse IDE.

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-video-recording/pom.xml

Yes, Xuggler is down, so depending on your needs you may consider moving to [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html), trying [JCodec](https://github.com/jcodec/jcodec) or [gstreamer-java](https://github.com/gstreamer-java).

mjoycemilburn, 2015-10-23:
Many thanks, Sarxos. There's a steep learning curve on all this stuff and
I'm still very much at the bottom of the slope.

The good new is that I've successfully run your motion-detection example
and have now started modifying it to suit my needs - so a big thank-you for
that.

Also, I've found a way /round/ my xuggler problem. I just wanted a clip to
use as a baseline for a speed-monitoring application. I realised I'd got a
package that came with my camera that would record an mpeg2 file, so I'm in
business on that front. Now all I have to do is tinker with your code so
that i read the video from my file instead of the camera - well it'll keep
me busy for a few days, I expect.

I shan't give up with xuggle completely though. It seems the best library
for the application and a real pity the authors couldn't make any money out
of it. I've got a book coming to help me get my head round Maven a bit
better - there's such a lot to learn and it's not very well-decribed on the
web.

I've noted your advice and I'll get back to you if I find an answer.

Best wishes, MartinJ

mjoycemilburn, 2015-11-01:
Hi again Sarxos. Now that I've tried the alternatives I've come back to
Webcam-camera and am determined to sort my Maven problems.

It seems that my problem with the Xuggler repository is a known deficiency
in Netbeans (supposedly fixed but apparently not). However this may be a
side issue as I see things a bit more clearly now and I think my build is
actually failing for other reasons - I've switched to Eclipse and am still
getting errors.

Right now I'm looking at an Eclipse error on your
webcam-capture/webcam-capture-examples/webcam
-video-recording/ pom where it says that line 6 has a "Project build error:
Non-resolvable parent POM: Could not find artifact
com.github.sarxos:webcam-capture-examples:pom:0.3.12-SNAPSHOT in xuggle
repo (http://xuggle.googlecode.com/svn/trunk/repo/share/
 java/) and 'parent.relativePath' points at wrong local POM".

I'm hoping this makes more sense to you than it does to me. My local
repositories include an empty Workspace Projects/com/github/sarxos. My
Central Repository has a com.github.sarxos with lots of jars but no
webcam-capture-examples pom. I feel I must have missed a step somewhere.

Hoping you can help me.

Best wishes, Martin J

mjoycemilburn, 2015-11-02:
Follow-up. Just to note. I think there's a spelling mistake on line 12 of
the webcam-capture-examples pom - missing n
in webcam-capture-example-video-recordig.

Unfortunately, fixing this doesn't make my problem disappear .

Regards, MJ

sarxos, 2015-11-02:
Hi @mjoycemilburn,

I designed Webcam Capture API Maven modularity in such a ways that all examples are integrated part of the whole project, but it may be hard to unlink them if someone is unfamiliar with Maven. The best way to work with examples is to import all Maven projects (modules), but if you are looking for a ready to use, standalone POM that does not depend on any parent project or SNAPSHOT version, here it is:

```xml
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
	<modelVersion>4.0.0</modelVersion>
	<groupId>com.github.mjoycemilburn</groupId>
	<artifactId>webcam-capture-example-video-recordig</artifactId>
	<version>1.0</version>
	<packaging>jar</packaging>
	<name>Webcam Capture - Video Recording Example</name>
	<description>Example how to record h264 video with awesome Xuggler library</description>
	<repositories>
		<repository>
			<id>xuggle repo</id>
			<url>http://xuggle.googlecode.com/svn/trunk/repo/share/java/</url>
		</repository>
	</repositories>
	<dependencies>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture</artifactId>
			<version>0.3.11</version>
		</dependency>
		<dependency>
			<groupId>xuggle</groupId>
			<artifactId>xuggle-xuggler</artifactId>
			<version>5.4</version>
		</dependency>
	</dependencies>
</project>
```

But in generally, to unlink project (module) from parent:

* remove ```<parent>```,
* specify your own ```<groupId>``` and ```<version>```, 
* use fixed version in dependency instead of ```${project.version}``` (the last stable is 0.3.11).

I tested it on my Ubuntu laptop and it works well generating ```output.ts``` file that I can open properly in [VLC](http://www.videolan.org/vlc/).

mjoycemilburn, 2015-11-03:
Many thanks Bartosz - that works a treat. Another donation to your site is
indicated.

​​However, I'm still trying to work out the implications of what you've
told me.

I think I see that the problem with my Build was that the pom for
com.github.sarxos:webcam-capture-examples:0.3.12-SNAPSHOT wasn't in my
local repository. I can just about infer from what you told me that this is
something to do with the "modules" entries in webcam-capture-examples.pom
and that this could be fixed by downloading more code. But here's my
question. Surely the whole point of Maven is to automate the Build process
and remove any need for "prepping" the project content?

I've got a wet towel round my head at present while I read "Introducing
Maven" (which I regretfully note doesn't get as far as "modules") and maybe
I'll eventually be able to answer my question for myself.

In the meantime, however, my project works and I can move forward again.
Also I don't have the problems I experienced with a failed attempt to use
OpenIMAJ, where the code wouldn't release the camera without me rebooting
my PC and where there is /no/ support. You might be interested to note,
however, that I had difficulties in both Webcam-Capture and OPenIMAJ with
buffer under-runs when trying to write an mpg file (I'm on Windows 10).
Switching to mp4 fixed this completely in Webcam-Capture.

Best wishes,

Martin J

sarxos, 2015-11-03:
Hi @mjoycemilburn,

Many thanks for the project donation :) 

When you take a look into the webcam-capture project root you may notice single ```pom.xml``` and some directories which are listen in pom in ```<modules>``` section. If you consider pom as a project, you can treat modules like a sub-projects. Webcam Capture API has many of them:

```plain
webcam-capture
├── webcam-capture
├── webcam-capture-addons
│   ├── webcam-capture-addon-spycam
│   └── webcam-capture-addon-swt
├── webcam-capture-drivers
│   ├── driver-ffmpeg-cli
│   ├── driver-fswebcam
│   ├── driver-gstreamer
│   ├── driver-ipcam
│   ├── driver-javacv
│   ├── driver-jmf
│   ├── driver-lti-civil
│   ├── driver-openimaj
│   ├── driver-v4l4j
│   ├── driver-vlcj
├── webcam-capture-examples
│   ├── webcam-capture-applet
│   ├── webcam-capture-detect-face
│   ├── webcam-capture-executable
│   ├── webcam-capture-javafx
│   ├── webcam-capture-javafx-fxml
│   ├── webcam-capture-live-streaming
│   ├── webcam-capture-manycams
│   ├── webcam-capture-motiondetector
│   ├── webcam-capture-onejar
│   ├── webcam-capture-painter
│   ├── webcam-capture-qrcode
│   ├── webcam-capture-swt-awt
│   ├── webcam-capture-transformer
│   ├── webcam-capture-video-recording
│   └── webcam-capture-websockets
└── webcam-capture-pages
```

When ```<parent>``` relationship is used, the child modules (projects) derive settings from the parent. This is really helpful, especially when the overall project structure grows in time, thus, many common settings can be put in a one place instead of keeping copies in every pom. Also, it's much easer to decouple dependencies and ```release:perform``` such structure into the Maven Central (less work has to be done because almost everything in such process can be automated). Maven very basics are pretty straightforward and easy to learn, but there are of course much more complex aspects like modules, thousands of different plugins, dependency management, inheritance, reporting, or plugin api (when you create your own maven plugins). The bigger your project is, the more subprojects you have, then, sooner or later you will find out that you need much more complex build management, and what I can tell is that Maven has never (8 years when I'm using it) disappointed me in this area - there is a bunch of plugins to cover almost every aspect of build process - from preparation, thru instrumentation, compilation, different kind of unit testing, various integration testing, reporting, packaging and release.

To clarify what are the implication of removing ```<parent>``` from this example pom. When you change the source code in webcam-capture-0.3.12-SNAPSHOT, and do ```mvn install``` (this install the build artifact in local repository) the change will not be visible in the project using webcam-capture-0.3.11, so if you want to play and hack the webcam-capture source code, it may be better to import complete project structure. But when it's enough to use stable version just stay with 0.3.11. This is the only implication. When new stable version is released, you just bump up 0.3.11 to 0.3.12 and voilla.

sarxos, 2015-11-17:
Hi @mjoycemilburn, 

Just FYI, I'm closing this ticket.

closed 2015-11-17 by sarxos

mjoycemilburn, 2015-11-17:
Thanks Sarxos. I'm sorry I've not replied before.

Because my Ant builds don't show the instability (which, incidentally isn't
as clear-cut as I indicated - it's now evident that re-boot doesn't always
clear the timeout) there's obviously something wrong with my Maven builds.
And because I can't exclude Netbeans from implication I'm now doing my
builds from the Windows command window.

So - i've been slowly pulling myself up by my own bootstraps, as it were,
and learning how to use Maven from the ground up. I've still to find a good
primer so it's been a slow process. So far I've been testing on my laptop
using the built-in camera. This has never shown the timeout problem so has
enabled me to concentrate on just getting maven working properly. Today I
switch to my desktop machine with an external USB camera and will start to
tackle the problem directly.

I'll let you know how I get on.

Regards, MartinJ

mjoycemilburn, 2015-11-21:
Right - things are becoming clearer. It now seems the problem is nothing to
do with Ant/Maven/Netbeans except insofar as the issues I was having with
my builds were obscuring the problem.

It's now evident that,whether I build with Ant or Maven, an execution of
the jar from the Windows command line on my Windows 7 machine regularly
fails with a camera timeout. But, frustratingly, it sometimes works. Worse
I can now no longer see any pattern to the failures (I previously thought a
reboot cleared the problem but not so).

However, the big news is that when I run the jar produced on the Windows 7
machine on my Windows 10 laptop everything is fine - at least I've tried on
numerous occasions now and it's worked every time. So it seems that the
timeout may be due to a setup problem on my Windows 7 (64 bit) machine

I've updated the camera drivers here to the latest versions but that hasn't
fixed things and the only other thing I can think of is to upgrade to
Windows 10.

But that's a big job - and before I launch into it I wonder if you can tell
me if /you're/ aware of any Windows 7 issues for webcam-capture?

Best wishes, MartinJ

sarxos, 2015-11-21:
Hi @mjoycemilburn,

Nope, I'm not aware of any Windows 7 specific issues, but this issue seems very strange. Have you tried building on a different PC that is also working on Windows 10?

Also, please make sure you are using 64-bit Java on 64-bin OS, same with 32-bit.

Can you please try enabling debug logs, so we have some better insight in what is happening on a BridJ level.

https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs

There will be a lot of stuff in the output, but maybe, just maybe, something may help dig the root cause and find out what the problem actually is (e.g. wrong lib loaded, capture format not supported, etc).

mjoycemilburn, 2015-11-22:
​
​Hi Sarxos - yes according to "java -d64 -version" I'm using 64 bit
Java on my 64 bit Windows 7 machine.

Everything I've built​ ​so far using webcam-capture​ ​on my Windows 10
machine works. Conversely all of​ ​the​ ​webcam-capture builds on the
Windows 7 machine show the instability.​ ​An OpenIMAJ build version of
the playvideoexample on Windows7
​ ​
is fine however.

The clincher for me is that the​ ​"jar-with-dependencies" for the
playvideo example
​ ​
produced on the Windows7 machine works fine on the Windows 10 machine.
So the instability has got to be something to do with a ​difference in
​the run-time environments - but something that's only triggered by
webcam-capture.

This is weird when the OpenImaj build is fine and uses the same
drivers as webcam-capture - right?

So, in the spirit of exploration and since I'm learning a lot through
all this floundering, I've tried to add the debugging you advised.

I've been getting debug logging all along  (that's why I know that the
camera is timing out) but I'm not sure I've got it fully configured so
followed your instructions

I put your logback.xml file alonside the pom.xml file in the program folder.
​ ​
I put the WebcamLogConfigurator call in the main class of the program.
The logback core and classic jars are already in the project as run
dependencies (they came in with xuggler and I presume this is why I've
been getting logging
​
​
) but I added them to my classpath anyway to my classpath - which I
assume is in directory target/classes)

However the debug output just looks the same as before - no
[capture-1] DEBUG lines, anyway. Have I done something wrong or is
this what you would expect?

Output from a run/file in Netbeans is shown below:

--- exec-maven-plugin:1.2.1:exec (default-cli) @ webcam-capture-play-video ---
Getting default webcam
16:21:57.923 [main] INFO  com.github.sarxos.webcam.Webcam -
WebcamDefaultDriver capture driver will be used
16:21:57.932 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:21:58.476 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
Opening webcam
16:21:58.481 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock
Webcam Conexant Polaris Video Capture 0
16:21:58.484 [atomic-processor-1] INFO
c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam Conexant Polaris
Video Capture 0
16:21:58.484 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device Conexant
Polaris Video Capture 0
16:21:58.484 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 0 starting session,
size java.awt.Dimension[width=176,height=144]
16:21:58.694 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
16:21:58.694 [atomic-processor-1] WARN
c.g.s.w.d.b.WebcamDefaultDevice - Different size obtained vs requested
- [176x144] vs [720x576]. Setting correct one. New size is [720x576]
16:21:58.697 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Initialize buffer
16:22:01.480 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:22:16.794 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Webcam device
com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@49462e6b is
now open
16:22:16.795 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is
now open Conexant Polaris Video Capture 0
16:22:16.805 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
16:22:19.808 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:22:21.931 [frames-refresher-[0]] ERROR
c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
16:22:21.940 [main] DEBUG c.github.sarxos.webcam.WebcamUpdater -
Webcam updater has been started
Instantiating buffered images
recording, please wait
16:22:21.944 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
16:22:24.946 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:22:26.973 [frames-refresher-[0]] ERROR
c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
16:22:26.980 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
16:22:29.980 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Closing webcam
16:22:31.954 [main] DEBUG com.github.sarxos.webcam.Webcam - Closing
webcam Conexant Polaris Video Capture 0
16:22:32.081 [frames-refresher-[0]] ERROR
c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
16:22:32.088 [atomic-processor-1] INFO
c.g.s.webcam.ds.cgt.WebcamCloseTask - Closing Conexant Polaris Video
Capture 0
16:22:32.088 [atomic-processor-1] DEBUG
c.g.s.w.d.b.WebcamDefaultDevice - Closing webcam device
16:22:32.089 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
16:22:32.287 [main] DEBUG c.github.sarxos.webcam.WebcamUpdater -
Webcam updater has been stopped
16:22:32.287 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock
Webcam Conexant Polaris Video Capture 0
16:22:32.288 [webcam-lock-[Conexant Polaris Video Capture 0]] DEBUG
com.github.sarxos.webcam.WebcamLock - Lock updater has been
interrupted
16:22:32.290 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam
Conexant Polaris Video Capture 0 has been closed
play
16:22:35.089 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:22:35.094 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0
16:22:38.095 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:22:38.101 [webcam-discovery-service] DEBUG
c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video
Capture 0