sarxos/webcam-capture #406: Recording Video From Multiple Cams using Multi Threading

MuhammadAmmad, 2015-11-03:
Hi @sarxos Thanks for such an amazing library. I m using video recording example with xuggler. It is running fine with one cam but when I add more cams (almost 13) then the recorded video is very slow. I m using threads. One thread for each cam but still it is slow. Plz help! Thanks again

sarxos, 2015-11-04:
Hi @MuhammadAmmad,

Did you perform some profiling (using e.g. ```jvisualvm```) to identify what code part is blocking? Are you using IP camera or UVC?

MuhammadAmmad, 2015-11-04:
Thanks @sarxos 
I m not doing any profiling.
Using ip Camera

sarxos, 2015-11-04:
Ok, can you profile your application with ```jvisualvm```? It's available in every JDK (Oracle, not OpenJDK).

In terms of IP camera, are you using PULL (JPEG) or PUSH (MPJEG)?

MuhammadAmmad, 2015-11-04:
OK i will try that
I m using PULL

sarxos, 2015-11-04:
How often do you get new image with pull? I'm asking because this may be a network issue. If you have ~13 cameras and fetching new image very very often, then transmitting all these JPEG images may clog your network interface or router/switch. If you have no delay between getting every new image, then I would suppose that this may be the root cause.

MuhammadAmmad, 2015-11-05:
I m using local IP
delay is 100ms

MuhammadAmmad, 2015-11-05:
public class WebcamViewerExample {

	static JSONObject config;
	

	public static void main(String[] args) throws MalformedURLException {
		Webcam.setDriver(new IpCamDriver(new IpCamStorage(args[0])));
		
		config=new JSONObject();
		try {
			 config=new JSONObject(readFromFile(args[1]));
					} catch (JSONException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
		for (Webcam webcam : Webcam.getWebcams()) {
			// recordVidio(webcam);
			new Capture(webcam).start();
			//break;
		}

		
	}

	public void recordVideo(Webcam webcam,String parentDir,int duration) {
		String camName = webcam.getName();
		//Date date=new Date(System.currentTimeMillis());
		Calendar calendar=Calendar.getInstance();
		
		String pathToSave=parentDir+"/"+camName+"/"+calendar.get(Calendar.YEAR)+"/"+calendar.get(Calendar.MONTH)+"/"+calendar.get(Calendar.DAY_OF_MONTH);
		String fileName=calendar.get(Calendar.HOUR_OF_DAY)+"-"+calendar.get(Calendar.MINUTE)+"-"+calendar.get(Calendar.SECOND);
		
		if(!createFolder(pathToSave)){
			System.out.println("Cant access path.");
			return;
		}
		File file = new File(pathToSave+"/"+fileName+".mp4");
		Dimension size = WebcamResolution.QVGA.getSize();

		webcam.setViewSize(size);
		webcam.open(true);

		long start = System.currentTimeMillis();
		ArrayList<VideoPic> bufferedImgs=new ArrayList<VideoPic>();
		for (int i = 0; i < duration*10; i++) {

			System.out.println(camName + " " + "Capture frame " + i);
			
			bufferedImgs.add(new VideoPic(webcam.getImage(), System.currentTimeMillis()));
			// 10 FPS
			try {
				Thread.sleep(100);
			} catch (Exception e) {
				// TODO Auto-generated catch block
				e.printStackTrace();
			}
		}

		//if(webcam.getName().equals("Cam 1"))
		new SaveClip(size, file, start, bufferedImgs).start();
		//recordVideo(webcam, parentDir,duration);
	}

	private void saveClip(Dimension size,File file, long start,ArrayList<VideoPic> arr) {
		
		IMediaWriter writer = ToolFactory.makeWriter(file.getAbsolutePath());
		writer.addVideoStream(0, 0, ICodec.ID.CODEC_ID_H264, size.width,
				size.height);
		
		for (int i = 0; i < arr.size(); i++) {
			VideoPic videoPic=arr.get(i);
		BufferedImage image = ConverterFactory.convertToType(
				videoPic.getBufferedImage(), BufferedImage.TYPE_3BYTE_BGR);
		IConverter converter = ConverterFactory.createConverter(image,
				IPixelFormat.Type.YUV420P);

		IVideoPicture frame = converter.toPicture(image,
				i * 100000);
		System.out.println("time-start: "+(videoPic.getTime() - start)+"");
		frame.setKeyFrame(i == 0);
		frame.setQuality(0);

		writer.encodeVideo(0, frame);
		}
		
		writer.close();
		System.out.println("Video recorded in file: " + file.getAbsolutePath());
	}

	private static final class Capture extends Thread {
		Webcam webcam;

		public Capture(Webcam webcam) {
			this.webcam = webcam;
		}

		@Override
		public void run() {
			new WebcamViewerExample().recordVideo(webcam,config.optString("path"),config.optInt("duration",600));
		}
	}
	
	private static final class SaveClip extends Thread {
		Dimension size;
		

		File file;
		long start;
		ArrayList<VideoPic> bufferedImgs;

		public SaveClip(Dimension size, File file, long start,
				ArrayList<VideoPic> bufferedImgs) {
			super();
			this.size = size;
			this.file = file;
			this.start = start;
			this.bufferedImgs = bufferedImgs;
		}
		@Override
		public void run() {
			new WebcamViewerExample().saveClip(size, file, start, bufferedImgs);
		}
	}

	private static boolean createFolder(String theFilePath) {
		boolean result = false;

		File directory = new File(theFilePath);

		if (directory.exists()) {
			System.out.println("Folder already exists");
			result=true;
		} else {
			result = directory.mkdirs();
		}

		return result;
	}
	
	static String readFromFile(String path){
		String line="";
		try {
            // FileReader reads text files in the default encoding.
            FileReader fileReader = new FileReader(path);
            // fileReader = new FileReader("config.txt");

            // Always wrap FileReader in BufferedReader.
            BufferedReader bufferedReader = 
                new BufferedReader(fileReader);

//            while((
            line = bufferedReader.readLine();
//            		) != null) {
                System.out.println(line);
           // }    

            // Always close files.
            bufferedReader.close();            
        }
        catch(FileNotFoundException ex) {
            System.out.println(
                "Unable to open file");                
        }
        catch(IOException ex) {
            System.out.println(
                "Error reading file");                   
            // Or we could just do this: 
            // ex.printStackTrace();
        }
	return line;
	}
	
	public class VideoPic{
		BufferedImage bufferedImage;
		long time;
		public VideoPic(BufferedImage bufferedImage, long time) {
			super();
			this.bufferedImage = bufferedImage;
			this.time = time;
		}
		public BufferedImage getBufferedImage() {
			return bufferedImage;
		}
		public void setBufferedImage(BufferedImage bufferedImage) {
			this.bufferedImage = bufferedImage;
		}
		public long getTime() {
			return time;
		}
		public void setTime(long time) {
			this.time = time;
		}
	}
}

MuhammadAmmad, 2015-11-05:
config.txt
{"path":"D:/Documents","duration":120}

MuhammadAmmad, 2015-11-16:
@sarxos any suggestions?

sarxos, 2015-11-16:
Hi @MuhammadAmmad ,

I tested your code pulling precaptured static images from Dasding web page (I have no access to so many IP cameras). The XML cameras descriptor I used can be found [here](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/resources/cameras.xml). Please note these images are updated once every 20 seconds, so this is not even close to 10 FPS, but I put some logs into the code and can clearly see that parallel fetching rate did not drop.

What I modified:

```java
System.out.println(System.currentTimeMillis() + " " + Thread.currentThread().getName() + " " + camName + " " + "Capture frame " + i);
bufferedImgs.add(new VideoPic(webcam.getImage(), System.currentTimeMillis()));
System.out.println(System.currentTimeMillis() + " " + Thread.currentThread().getName() + " " + camName + " " + "Capture frame " + i + " done");
```

And the output:

```
timestamp     thread   camera     msg
1447666351817 Thread-4 Dasding 06 Capture frame 0
1447666351818 Thread-4 Dasding 06 Capture frame 0 done
1447666351844 Thread-5 Dasding 07 Capture frame 0
1447666351844 Thread-5 Dasding 07 Capture frame 0 done
1447666351852 Thread-6 Dasding 10 Capture frame 0
1447666351852 Thread-6 Dasding 10 Capture frame 0 done
1447666351918 Thread-4 Dasding 06 Capture frame 1
1447666351918 Thread-4 Dasding 06 Capture frame 1 done
1447666351920 Thread-3 Dasding 04 Capture frame 0
1447666351920 Thread-3 Dasding 04 Capture frame 0 done
1447666351939 Thread-2 Dasding 02 Capture frame 0
1447666351939 Thread-2 Dasding 02 Capture frame 0 done
1447666351944 Thread-5 Dasding 07 Capture frame 1
1447666351945 Thread-5 Dasding 07 Capture frame 1 done
1447666351952 Thread-6 Dasding 10 Capture frame 1
1447666351952 Thread-6 Dasding 10 Capture frame 1 done
1447666352018 Thread-4 Dasding 06 Capture frame 2
1447666352018 Thread-4 Dasding 06 Capture frame 2 done
1447666352020 Thread-3 Dasding 04 Capture frame 1
1447666352020 Thread-3 Dasding 04 Capture frame 1 done
1447666352045 Thread-2 Dasding 02 Capture frame 1
1447666352045 Thread-5 Dasding 07 Capture frame 2
1447666352045 Thread-2 Dasding 02 Capture frame 1 done
1447666352045 Thread-5 Dasding 07 Capture frame 2 done
1447666352052 Thread-6 Dasding 10 Capture frame 2
1447666352052 Thread-6 Dasding 10 Capture frame 2 done
1447666352054 Thread-1 Dasding 01 Capture frame 0
1447666352054 Thread-1 Dasding 01 Capture frame 0 done
1447666352118 Thread-4 Dasding 06 Capture frame 3
1447666352118 Thread-4 Dasding 06 Capture frame 3 done
1447666352121 Thread-3 Dasding 04 Capture frame 2
1447666352121 Thread-3 Dasding 04 Capture frame 2 done
1447666352145 Thread-2 Dasding 02 Capture frame 2
1447666352145 Thread-5 Dasding 07 Capture frame 3
1447666352145 Thread-5 Dasding 07 Capture frame 3 done
1447666352145 Thread-2 Dasding 02 Capture frame 2 done
1447666352152 Thread-6 Dasding 10 Capture frame 3
1447666352152 Thread-6 Dasding 10 Capture frame 3 done
1447666352154 Thread-1 Dasding 01 Capture frame 1
1447666352154 Thread-1 Dasding 01 Capture frame 1 done
... etc
```

Therefore, if there is a lag in fetching images through the network, it's may be caused by either lag on networking side or in IP camera, but there is no issue in the Webcam Capture API.

For example, camera Dasding 01 was executed every 100 ms, with quite good precision:

```
$  cat times.txt | grep "Dasding 01" | grep done
```

Gives:

```
[... 95 lines cut ...]
1447666361818 Thread-1 Dasding 01 Capture frame 95 done
1447666361918 Thread-1 Dasding 01 Capture frame 96 done
1447666362018 Thread-1 Dasding 01 Capture frame 97 done
1447666362118 Thread-1 Dasding 01 Capture frame 98 done
1447666362219 Thread-1 Dasding 01 Capture frame 99 done
```

Just FYI, your code threw an error on writing images to files:

```plain
Exception in thread "Thread-20" java.lang.RuntimeException: could not open stream com.xuggle.xuggler.IStream@-1409148128[index:0;id:0;streamcoder:com.xuggle.xuggler.IStreamCoder@-1409147856[codec=com.xuggle.xuggler.ICodec@-1409148464[type=CODEC_TYPE_VIDEO;id=CODEC_ID_H264;name=libx264;];time base=1/1000000;frame rate=0/0;pixel type=YUV420P;width=770;height=433;];framerate:0/0;timebase:1/90000;direction:OUTBOUND;]: Operation not permitted
	at com.xuggle.mediatool.MediaWriter.openStream(MediaWriter.java:1192)
	at com.xuggle.mediatool.MediaWriter.getStream(MediaWriter.java:1052)
	at com.xuggle.mediatool.MediaWriter.encodeVideo(MediaWriter.java:742)
	at com.xuggle.mediatool.MediaWriter.encodeVideo(MediaWriter.java:783)
	at com.github.sarxos.webcam.WebcamViewerExample.saveClip(WebcamViewerExample.java:91)
	at com.github.sarxos.webcam.WebcamViewerExample.access$0(WebcamViewerExample.java:76)
	at com.github.sarxos.webcam.WebcamViewerExample$SaveClip.run(WebcamViewerExample.java:130)
```

I haven't investigate this further due to my time limits, but I see no issue in the code in Webcam Capture API code. I suppose your problem may be either:

* Throttling in networking,
* IP camera is not able to generate so many FPS per second in PULL mode,
* Xuggler issue with parallel stream recording (this is out of scope of my knowledge).

MuhammadAmmad, 2015-11-16:
OK Thanks alot.

closed 2015-11-16 by MuhammadAmmad

sachinrke, 2018-10-08:
hi @MuhammadAmmad ,
Thanks for the code. I have also tried your code and able to record video from multiple cameras. but this code not recording audio with video.so how can we record audio with video together from multiple cameras. please help me.
 i will be very grateful to you.

Thank You 
sachin pal