sarxos/webcam-capture #407: Streaming video

swap17, 2015-11-04:
I need to implement video streaming using applet.The functionality is that i need to connect to remote user,how is using camera and i need to watch the user activity.Please Suggest how may i start with integration.

sarxos, 2015-11-04:
Hi @swap17,

I would suggest to **not use applets at all**. This functionality is almost dead - vendors (like Oracle, Google, etc) are doing things to keep it in the grave:
* Self-signing [does not work any more](https://blogs.oracle.com/java-platform-group/entry/new_security_requirements_for_rias) on clients having Java 7u51+, so your clients will not run Applets that are not signed by real certificate that you bought (you have to put money into certificate, every year),
* You will not run applets on Chrome because Google [disabled](https://java.com/en/download/faq/chrome.xml) NPAPI.
* You will run applet with Webcam Capture API, but there are some small, but noisy issues, that you still have to keep in mind when developing applets (e.g. applet restart which is not really restart).

Please consider using WebRTC features that was designed for this sole purpose (instead of applets). Take a look at this demo:

https://www.webrtc-experiment.com/broadcast/

Start broadcast on the PC, open the same page on the other PC, and join the broadcast room (by providing some unique name). Tested between Firefox and Chrome and works well. I guess this is what you are looking for. Here is the table demonstrating what WebRTC features are working on a different browsers:

http://iswebrtcreadyyet.com/

Safari is retarded (like always), so you may consider to use applet only in case of this browser, this is up to you, but in my opinion applets will remain dead functionality probably forever and more and more browsers will be disabling it in the future.

If you are still willing to stick into the applet, then here is the working demo, but you have to change your Java security settings to allow running self-signed code in case you are running Java 7u51 or newer.

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-applet

I strongly suggest to use Maven to build this example (signing JARs manually is a total pain). Please let me know if you have problems with that, I will change the pom so it's not linked to the parent.

swap17, 2015-11-04:
Thanks for suggestion sarxos.I need to support IE and safari which is not possible using webRTC so i need to go with applet.Also i have implemented applet for capturing image in my project.Now i need to implement video streaming using applet.So may me provide me more clear way to implement with applet.

sarxos, 2015-11-04:
Hi @swap17,

I get your point. Consider using Maven to build the applet because building and signing it manually is a total pain. You may also try Gradle or ant.

Just copy the code from the example and change to fit your needs:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-applet

And replace ```pom.xml``` with the one provided below. Please note it's using some test ``` keystore.jks``` with password _test1234_, but if you want it to run on your clients without a need of reconfiguring Java security settings, you will need to import trusted certificate. I cannot give you any details how this certificate should be imported into JKS since I never bought one and I was always using self-signing.

New ```pom.xml```:

```xml
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
	<modelVersion>4.0.0</modelVersion>
	<groupId>com.github.swap17</groupId>
	<artifactId>webcam-capture-example-applet</artifactId>
	<version>1.0</version>
	<packaging>jar</packaging>
	<name>Webcam Capture - Applet Example</name>
	<description>Example of how to use Webcam Capture within Java Applet</description>
	<dependencies>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture</artifactId>
			<version>0.3.11</version>
		</dependency>
	</dependencies>
	<build>
		<finalName>applet</finalName>
		<plugins>
			<plugin>
				<artifactId>maven-compiler-plugin</artifactId>
				<version>2.3.2</version>
				<configuration>
					<source>1.6</source>
					<target>1.6</target>
				</configuration>
			</plugin>
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-resources-plugin</artifactId>
				<version>2.6</version>
				<executions>
					<execution>
						<phase>prepare-package</phase>
						<goals>
							<goal>copy-resources</goal>
						</goals>
						<configuration>
							<outputDirectory>${project.build.directory}</outputDirectory>
							<overwrite>true</overwrite>
							<resources>
								<resource>
									<directory>src/main/html</directory>
									<includes>
										<include>**/*.html</include>
									</includes>
									<filtering>true</filtering>
								</resource>
							</resources>
						</configuration>
					</execution>
				</executions>
			</plugin>
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-shade-plugin</artifactId>
				<version>2.3</version>
				<configuration>
				<artifactSet>
					<includes>
						<include>*:*</include>
					</includes>
				</artifactSet>
				</configuration>
				<executions>
					<execution>
						<phase>package</phase>
						<goals>
							<goal>shade</goal>
						</goals>
					</execution>
				</executions>
			</plugin>
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-jarsigner-plugin</artifactId>
				<version>1.3.2</version>
				<configuration>
					<keystore>keystore.jks</keystore>
					<alias>example</alias>
					<storepass>test1234</storepass>
					<keypass>test1234</keypass>
				</configuration>
				<executions>
					<execution>
						<phase>package</phase>
						<goals>
							<goal>sign</goal>
						</goals>
					</execution>
				</executions>
			</plugin>
		</plugins>
	</build>
</project>
```

After you replace the ```pom.xml```, just go to the directory where it's located and run:

```plain
mvn clean install
```

The above will rebuild the project from scratch. You will find your artifacts in ```target``` directory.

For more details on how to use Maven you can google a lot of useful resources. It's one of the most common build managers ever used by Java community.

swap17, 2015-11-06:
Thanks for your suggestion.Also is there a way in webcam library for screen monitoring.

sarxos, 2015-11-09:
@swap17, by now there is no way to monitoring screen.

closed 2015-11-12 by sarxos