sarxos/webcam-capture #408: Applications which uses webcam capture freezes at end or gives a error

JuhaMcEla, 2015-11-05:
I have Raspberry Pi Model B+ and Rasbian Jessie OS.

When I try to execute How to take picture and save to file -example:
<pre><code>import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

public class TakePictureExample {
    
  
      static {

        System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "trace");
    }
    
    static {
        Webcam.setDriver(new V4l4jDriver()); // this is important
    }
      
    public static void main(String[] args) throws IOException {

        // get default webcam and open it
        Webcam webcam = Webcam.getDefault();
        webcam.open();

        // get image
        BufferedImage image = webcam.getImage();

        // save image to PNG file
        ImageIO.write(image, "PNG", new File("test.png"));
    }
}

</code></pre>

It gives my this error:
<pre><code>
[main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver V4l4jDriver
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver - Video file detected /dev/video0
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Creating V4L4J device from file /dev/video0
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Get video device info
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support BGR conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support JPG conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support RGB conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support YUV conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Support YVU conversion true
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing MJPEG (DISCRETE)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing RGB24 (UNSUPPORTED)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing BGR24 (UNSUPPORTED)
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing YUV420 (UNSUPPORTED)
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Best image format match MJPEG - 48
[main] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver V4l4jDriver does not support this feature
[main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam /dev/video0
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam /dev/video0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Opening V4L4J device /dev/video0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Constructing V4L4J frame grabber
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Starting V4L4J frame grabber
[pool-1-thread-1 - v4l4j push source] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Next frame au.edu.jcu.v4l4j.JPEGVideoFrame@1bbd988
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Webcam V4L4J is now open
[main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open /dev/video0
[pool-1-thread-1 - v4l4j push source] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Next frame au.edu.jcu.v4l4j.JPEGVideoFrame@9f6fc5
java.lang.IllegalArgumentException: image == null!
	at javax.imageio.ImageTypeSpecifier.createFromRenderedImage(ImageTypeSpecifier.java:925)
	at javax.imageio.ImageIO.getWriter(ImageIO.java:1592)
	at javax.imageio.ImageIO.write(ImageIO.java:1520)
	at TakePictureExample.main(TakePictureExample.java:40)
</code></pre>

Sometimes application runs at the end and takes a picture, but then it freezes. 

Jars I have included:
<ul>
<li>slf4j-api-1.7.2</li>
<li>slf4j-simple-1.7.2</li>
<li>v4l4j-0.9.1-r507</li>
<li>webcam-capture-0.3.11</li>
<li>webcam-capture-driver-v4l4j-0.3.11</li>
</ul>

sarxos, 2015-11-12:
Hi @JuhaMcEla,

Thank you for the report. I will be investigating this. It may be linked to #382.

JuhaMcEla, 2015-11-12:
Thank you for your the reply @sarxos.

About that linked issue you refered, your examples (and I have tried):

<ul>
<li>How to take picture and save to file</li>
<li>How to record video from webcam</li>
</ul>

freezes at the end on me, regardless if I use raspicam or standard USB webcam (Microsoft LifeCam VX-7000) and I can't run those examples even for the first time.