sarxos/webcam-capture #409: Timeout on Conexant Polaris camera

mjoycemilburn, 2015-11-08:
Building this example using Maven on Netbeans is delivering inconsistent results. Sometimes the project runs successfully but at others it just delivers a blank screen. Debugging info suggests the camera is timing out.

An Ant build is fine. The camera is fine. I can find no pattern to the failures 

I'm doing this on 64-bit Windows 10 and can't understand how the Maven build works out that it needs the 64 bit version  of the openIMAJ library or how it gets the natives. I've tried forcing the situation by including a dependency on openIMAJ and including a setDriver call, but this still produces a blank screen (plus different errors). 

Attached typical output from a failed run. I think something is wrong at a pretty subtle level as sometimes I get program exceptions too.

------------------------------------

cd C:\Users\Administrator\Documents\NetBeansProjects\WebcamCapturePlayVideo; "JAVA_HOME=C:\\Program Files\\Java\\jdk1.8.0_20" cmd /c "\"\"C:\\Program Files\\NetBeans 8.0.1\\java\\maven\\bin\\mvn.bat\" -Dexec.args=\"-classpath %classpath com.github.mjoycemilburn.webcamcaptureplayvideo.PlayVideoFromCamera\" -Dexec.executable=\"C:\\Program Files\\Java\\jdk1.8.0_20\\bin\\java.exe\" -Dexec.classpathScope=runtime -Dmaven.ext.class.path=\"C:\\Program Files\\NetBeans 8.0.1\\java\\maven-nblib\\netbeans-eventspy.jar\" org.codehaus.mojo:exec-maven-plugin:1.2.1:exec\""
Running NetBeans Compile On Save execution. Phase execution is skipped and output directories of dependency projects (with Compile on Save turned on) will be used instead of their jar artifacts.
Scanning for projects...
                                                                        
------------------------------------------------------------------------
Building Webcam Capture - Play Video from Camera Example 1.0
------------------------------------------------------------------------

--- exec-maven-plugin:1.2.1:exec (default-cli) @ webcam-capture-play-video ---
Getting default webcam
11:52:33.762 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
11:52:33.772 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:52:34.208 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
Opening webcam
11:52:34.212 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Conexant Polaris Video Capture 0
11:52:34.218 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam Conexant Polaris Video Capture 0
11:52:34.218 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device Conexant Polaris Video Capture 0
11:52:34.218 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=176,height=144]
11:52:34.367 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
11:52:34.367 [atomic-processor-1] WARN  c.g.s.w.d.b.WebcamDefaultDevice - Different size obtained vs requested - [176x144] vs [720x576]. Setting correct one. New size is [720x576]
11:52:34.370 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Initialize buffer
11:52:37.232 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:52:52.687 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@690a3c55 is now open
11:52:52.692 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Conexant Polaris Video Capture 0
11:52:52.698 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:52:55.699 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:52:57.787 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
11:52:57.802 [main] DEBUG c.github.sarxos.webcam.WebcamUpdater - Webcam updater has been started
Instantiating buffered images
recording, please wait
11:52:57.808 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:53:00.809 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:53:02.873 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
11:53:02.880 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:53:05.880 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Closing webcam
11:53:07.820 [main] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam Conexant Polaris Video Capture 0
11:53:07.962 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
11:53:07.968 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamCloseTask - Closing Conexant Polaris Video Capture 0
11:53:07.969 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Closing webcam device
11:53:07.970 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:53:08.166 [main] DEBUG c.github.sarxos.webcam.WebcamUpdater - Webcam updater has been stopped
11:53:08.166 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam Conexant Polaris Video Capture 0
11:53:08.170 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam Conexant Polaris Video Capture 0 has been closed
play
11:53:08.172 [webcam-lock-[Conexant Polaris Video Capture 0]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
11:53:10.970 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:53:10.977 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:53:13.977 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:53:13.982 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
11:53:16.984 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:53:16.989 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Conexant Polaris Video Capture 0
------------------------------------------------------------------------
BUILD SUCCESS
------------------------------------------------------------------------
Total time: 46.055s
Finished at: Sun Nov 08 11:53:17 GMT 2015
Final Memory: 11M/93M
------------------------------------------------------------------------

sarxos, 2015-11-09:
This is just my first guess, but if project build with one tool fails to run properly, and build with a different tool works well, then I would assume that dependencies for these two builds are different. Therefore, can you please make sure that you are using the same JARs in both Ant and Maven build?

mjoycemilburn, 2015-11-09:
I only mentioned the Ant build success to give assurance that my camera and
its supporting USB drivers are sound. It's the instability of my Maven
build that I want to concentrate on.

I've just rerun my Maven build (nothing changed since yesterday) after a
morning off to clear my head and it works. But if I repeat the test (ie
submit File Run again) the camera times out again and I just get a black
image. It seems a reboot and patience while Netbeans loads clears the
problem. It's got to be some sort of weird memory/unreleased thread issue.

My pom is as below:

<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="
http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="
http://maven.apache.org/POM/4.0.0
http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>
    <groupId>com.github.mjoycemilburn</groupId>
    <artifactId>webcam-take-picture</artifactId>
    <version>1.0</version>
    <packaging>jar</packaging>
    <name>Webcam Capture - Take Picture Example</name>
    <description>Example how to create pic file with awesome Xuggler
library</description>
    <repositories>
        <repository>
            <id>xuggle repo</id>
            <url>http://xuggle.googlecode.com/svn/trunk/repo/share/java/
</url>
        </repository>
    </repositories>
    <dependencies>
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.11</version>
        </dependency>
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture-driver-openimaj</artifactId>
            <version>0.3.11</version>
        </dependency>
        <dependency>
            <groupId>xuggle</groupId>
            <artifactId>xuggle-xuggler</artifactId>
            <version>5.4</version>
        </dependency>
    </dependencies>
</project>

Does the pom look OK to you? I don't really need the xuggle  references I
guess but these have the effect of turning on debug logging so I've kept
them. I may not need the openimaj driver dependency either since I'm pretty
sure the basic openimaj driver will do the job and I'm not currently
calling setdriver in my java code.

When I expand the Dependencies folder in my project I can see the openIMAJ
drivers inside my webcam-capture-0.3.11 jar for both win32 and win64.  I
can't see how webcam-capture will know which one to use, but then this may
not be the problem. Any more ideas?  Maybe I should try in Eclipse?

Regards, Martin J

PS I now see I've got the same problem in Netbeans on the very simplest
Capture Image example

sarxos, 2015-11-10:
Hi @mjoycemilburn,

I'm not very familiar with Netbeans and how it integrates with Maven (I'm using Eclipse), but what you said leads me to the conclusion that your previous run may not be actually completed correctly. For example, when you are invoking ```mvn exec``` do you know how Maven/Netbeans close your application? If application is stopped with KILL signal, then, most likely, the memory is not cleaned after the run, thus making these strange problems. Pretty same behaviour, sometimes with random crashes, may be observed when you run your application from Eclipse, and then kill it. The next run may, but does not have to succeed, and usually it's done, runtime is wasted and one has to restart PC to get it clean. This is observed only on Windows. Linux and Mac are not affected by this problem.

How to verify:

* Start your app and stop with the graceful approach (e.g. invoke ```System.exit(0)``` somewhere in your application after you press the button or something). Do that few times. Everything should be working well.
* Start your app with the same way as you did before. If previous experiment succeeded, it's most probably the case.

How to workaround:

* The **most correct way** is to close application properly, or
* Use ```SignalHandler``` to intercept OS signals and close the application properly (this is ugly solution and must not be used in production code),
* Use ```Webcam.setHandleTermSignal(true)``` to use build-in hax (but this solution is also ugly since it uses the above method).

From the doc:

> TERM signal handler is a development utility and shall not be used in production. The goal of this feature is to handle application termination to gracefully close all open webcams. This has been done to avoid termination issues existing on some operating systems. For example, when running on Windows and terminated in specific moment, native layer can become broken and user can observe very strange problems like e.g. no webcams detected, impossible to open capture device or to close it properly. The complete spectrum of unwanted side effects is pretty long, but to avoid all of them it’s enough to gracefully close JVM so all webcam shutdown hooks are fired, or to enable TERM signal which will do that instead.

> To enable TERM signal handling one has to call below static method:

> ```java
> Webcam.setHandleTermSignal(true);
> ```

> Same to disable it:

> ```java
> Webcam.setHandleTermSignal(false);
> ```

> It’s very important to set this flag before any webcam has been open or it will not have any effect.

> Example:
> ```java
> public class Test {
> 
>     static {
>         Webcam.setHandleTermSignal(true);
>     }
> 
>     public static void main(String args[]) {
>         Webcam w = Webcam.getDeafult();
>         w.open();
>         // do something here and terminate application in the middle
>     }
> }
> ```

> TERM signal handler should be used in non-production code only. It has been designed to workaround Windows issues arising when one is killing running application when webcam is open (e.g. JVM crash, problems with camera detection, etc). In most cases, when TERM signal handler is not used, it's enough to restart Windows to work with webcam device same as before. 

Please let me know if above fixes your issue.