sarxos/webcam-capture #411: Is there a way to set the frame rate?

timmolter, 2015-11-10:
I'd like to capture at 10 frames per second. Can I set that somewhere? Thanks in advance!

timmolter, 2015-11-11:
I think I found it:

        boolean started = grabber.startSession(size.width, size.height, 50, Pointer.pointerTo(device));

in `WebcamDefaultDevice`.

So I reduced it a bit, and observed no reduction in CPU usage. One CPU at 100% still even when capturing at 1 frame per second. This leads me to question how image capture works below the hood. Can you provide any insight?

sarxos, 2015-11-12:
Hi @timmolter,

The setting you are referring to in ```WebcamDefaultDevice``` specifies maximum FPS, but from what I remember it's valid only on Mac OS. The other systems does not regulate FPS by any value. In general the FPS is controlled by the camera hardware and there is no _easy_ way to change it.

If you need very accurate timing, when you cannot afford waiting for a frame, you may open camera in async mode (```webcam.open(true)```) and read image every 100ms, but please keep in mind that when camera is not able to fetch new image in this 100ms, you will get old image instead.

In regards to CPU consumption. I'm not sure if you know, but Webcam Capture API is sending requests to fetch new image even when you don't need it so you are not impacted by the webcam buffer delay, especially annoying with low FPS. This fetching causes 30% CPU consumption on my PC. When you use ```WebcamPanel``` to render image, this value grows to 50% or even 100% CPU when window is fullscreen (due to intensive rendering demands).

You may try to profile your application with e.g. ```jvisualvm``` to find out what code is consuming your CPU.

[image]

From what I did on my PC I can see that most time consuming task are:

* ```AtomicProcessor.process(..)``` - let's discard it. It's time consuming because it's blocking, not because there is so intensive CPU load.
* ```RepaintManager$ProcessingRunnable.run()``` - this is Swing painting, it probably can be boosted by using ```VolatileImage``` but I'm not a Swing expert, so any contribution here is highly appreciated if anyone knows how to fix painting performance.

timmolter, 2015-11-13:
@sarxos Thanks, I appreciate you thorough answer. After digging into your code, pulling out just the essentials, and removing the Swing GUI, the CPU usage is more on the order of 40%, which is much more acceptable for my purposes. I ended up creating a new project with your source code and some of openimaj's in order to get access to the raw btyes coming from the camera. While doing so I found a place in openimaj's code where the frame rate is hardcoded, and I'll see if changing that will affect things.

BTW, you should update your openimaj dependency to the latest version. On a Mac, your examples using the openimaj didn't work for me, but when I increased the openimaj version it did work.

closed 2015-11-13 by timmolter

sarxos, 2015-11-13:
@timmolter, thank you for the hint (I will bump up OpenIMAJ version) and good luck with your project.