sarxos/webcam-capture #412: Error Webcam Capture

yfabio, 2015-11-13:
Hi,

I've been tried this little piece of code, but I got an error. 

public class App {
	public static void main(String[] args) {
		Webcam webcam = Webcam.getDefault();
		webcam.open();
		Path dir = Paths.get("text.jpg");
		try {
			ImageIO.write(webcam.getImage(), "test", Files.newOutputStream(dir));
		} catch (IOException e) {
			e.printStackTrace();
		} finally {
			webcam.close();
		}
	}
}

error

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

I set up all the configuration on maven correct but when I run that code above nothing happen. 

What do I suppose to do ?

sarxos, 2015-11-13:
Hi @yfabio. Thank you for the report. This happens because you haven't added any SLF4J logger implementation into the classpath. For more details please see URL that is mentioned in this warning: http://www.slf4j.org/codes.html#StaticLoggerBinder

closed 2015-11-13 by sarxos

yfabio, 2015-11-13:
Ok Thank you. I'll try again.

sarxos, 2015-11-14:
Just for our information. This will not cause any problems except the fact that you wont see any logs. It depends if you need them or not.

sarxos, 2015-11-14:
Ah, in regards to you statement:

>  when I run that code above nothing happen

Can you please elaborate a little more? Is there a _test_ image created in a project directory? Is webcam detected?