sarxos/webcam-capture #417: Webcam  screen is black

arumbala72, 2015-11-16:
Hi Sarxos,
First I should appreciate the work you have done for webcam features in java ,I have installed your software webcam capture 3.0 and it was working fine on windows 8 but suddenly it stop working i do not know why ? I have done windows update and java update otherwise nothing changed and it is same webcame and now it is started giving the below error and it has deducted but it giving error like request time out when getting image from webcam
Exception in thread "AWT-EventQueue-0" com.github.sarxos.webcam.WebcamException: Cannot execute task

My OS is Windows 8.1 pro on oracle virtual box  with MAC OS X and my camera is working fine when i do Skype.

Please help me.

Thank you ,
arum

sarxos, 2015-11-16:
Hi @arumbala72,

What is the error exactly - please provide it with full stack trace. This one line is meaningless because it does not tell us which task cannot be executed.

Does it happen after you restart your PC?

Just FYI, Skype has its own API to access camera.

arumbala72, 2015-11-18:
Hi Sarxos,
I am sorry ,i have been out of work couple of days just seen your email
 and this is the error i am getting ,it was working fine for 8 months and
suddenly stop working..

System  is  Windows 8.1 pro  on virtual box  with  MAC  OS X  EICapiten

You help would be highly appreciable.

Thanks
arum

log4j:WARN No appenders could be found for logger
(com.github.sarxos.webcam.Webcam).
log4j:WARN Please initialize the log4j system properly.
size.start() java.awt.Dimension[width=176,height=144]
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot
execute task
at
com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
at
com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
at
com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
at com.github.sarxos.webcam.WebcamPanel.start(WebcamPanel.java:796)
at
com.medrecordonline.utilities.TakeSnapshotFromSystemCamera.<init>(TakeSnapshotFromSystemCamera.java:196)
at
com.medrecordonline.utilities.TakeSnapshotFromSystemCamera.main(TakeSnapshotFromSystemCamera.java:443)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native
grabber!
at
com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:330)
at
com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
at
com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
at
java.util.concurrent.ThreadPoolExecutor$Worker.runTask(ThreadPoolExecutor.java:886)
at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:908)
at java.lang.Thread.run(Thread.java:662)

sarxos, 2015-11-18:
The webcam cannot be open. You may enable debug logging to have better insight in what is happening. https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs

arumbala72, 2015-11-20:
Bartosz,
Thank you for your help and valuable time .
I have tried debug mode and i found the issue ,image request time out and i
changed different camera ,it is works fine.
Do you know to record audio with video ? i have seen ,encoder.java,which
records only video but i do need to record both audio and video together.
Is there any source available?

Thanks,
arum

sarxos, 2015-11-20:
Hi @arumbala72,

I'm not aware of any ready to use example demonstrating how to record audio and video using Webcam Capture API, but I'm sure this is possible using Xuggler.

http://stackoverflow.com/questions/14029555/targetdataline-and-xuggler-to-record-audio-with-a-video-of-the-screen

_(see the responses where ```writer.encodeAudio(..)``` is used to add audio to the recording)_

I never tried accessing audio, but you may take a look at this code:

https://github.com/ryancheu/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-live-streaming/src/main/java/us/sosia/video/stream/agent/StreamServerAgent.java

Especially inner class ```AudioGrabTask```.

arumbala72, 2015-11-21:
Thank you Bartosz,the work you have done for community  is amazing and well
done.

arumbala72, 2016-02-24:
Bartosz,
Hope you are doing well and I have asked you some queries about video
recording and now i have successfully synced video and audio together using
xuggler and your library  but there is delay in video  ,i can see the lips
delay...however when i load webcam with your code,i noticed the delay in
image moving and compare with Skype same camera,there is delay in your
webcam ,how do i rectify that?,i think some images missing between frames
when capturing and I found WebcamUpdater class ,i do not know use of
this class,will it solve this issue?

Your help would be very much appreciated.

Thank you.
arum

sarxos, 2016-02-25:
@arumbala72,

Image from Webcam Capture API can be delayed because it needs to go through the native and Java code, when Skype use only native code. Skype also operates on much lower level and has huge room to boost performance. However, I think delay in Webcam Capture API may be tens of milliseconds at most.

The ```WebcamUpdater``` is internal class and should not be used directly. Its goal is to constantly update webcam buffer so the image is always the newest one. Let me describe the reason of its existence. From what I found (using OpenIMAJ) the webcam buffer consist of 5 elements and acts as a queue. Whenever new frame is fetched it is placed on the queue tail, and user asking for a new image is always getting element from queue head. The ```WebcamUpdater``` has been designed to minimize the influence of the buffer existence (to minimize delay) - it constantly ask webcam to fetch new frame, so buffer is always up to date. This is the best I was able to do to get rid of delay caused by this feature.

In regards to the timing - if you really need it to be very accurate, and you cannot accept blocking, you can open the webcam in async mode. In normal use cases webcam is open in blocking mode, so when you call ```getImage()``` the call will block until new image is received by the API (this usually takes 33ms if you assume 30 FPS, and more time when FPS is lower). The async mode, in opposite, does not wait for the image to be received but returns the one which was fetched previously. The downside is the fact that you may get the same image twice, or more, if you call ```getImage()``` with frequency higher than actual camera FPS.

To open webcam in async mode it's enough to pass ```true``` to the ```open()``` method, like this:

```java
webcam.open(true);
```

I'm not sure if this may help you resolving the problem, but it's still worth to try :)

arumbala72, 2016-02-25:
Thank you Bartosz and now i understand more the logic behind your code and I will try in async mode and let you know the feedback.

I noticed this delay only when i play back recoding with audio after merging video and audio file together  using xuggler .

I really appreciate your work and prompt reply email ,you are very helpful.

Thank you.

arum

arumbala72, 2017-05-04:
Hi Bartosz,
Hope you are doing well.
I would like to grab your brain ,is there any chance we can capture image in java using capture button in webcam instead of having our own take picture button?

Thanks,
Arum

sarxos, 2017-05-10:
Hi @arumbala72,

This is not possible with this library only. I think you will have to reverse-engineer your camera USB protocol and implement your own trigger using usb4java, libusbjava or equivalent.

You can find some more or less usable instructions in Andrew Davison's book, NUI chapter 4:

http://fivedots.coe.psu.ac.th/~ad/jg/nui04/launcher.pdf

Regards

arumbala72, 2017-05-10:
Thanks Bartosz,I will check it.

arumbala72, 2017-09-08:
Hi Bartosz,
Hope you are doing well.I have requirement to capture images minimum 3
megapixels ,i have tried below code but some reason it captures only 1280 *
960 but in video motion i can see QXGA(2048, 1536)  this size,do you know
what i am doing wrong?

        Dimension[] nonStandardResolutions =
          new Dimension[] { WebcamResolution.PAL.getSize(),
                            WebcamResolution.QXGA.getSize(),
                            new Dimension(2000, 1000),
                            new Dimension(1000, 500), };
        //      Dimension[] nonStandardResolutions =
        //        new Dimension[] { WebcamResolution.UXGA.getSize() };
        //        Dimension d = new Dimension(2048, 1536);
        //        webcam.setCustomViewSizes(new Dimension[] { d });
        //        webcam.setViewSize(d);
        webcam.setCustomViewSizes(nonStandardResolutions);
        webcam.setViewSize(WebcamResolution.QXGA.getSize());
        Dimension size = WebcamResolution.QXGA.getSize(); // 640 x 480
        liveImagePanel = new WebcamPanel(webcam,size, false);

  bufferedimage = webcam.getImage();

    Tools.debug(" * * * Inside Capture 1 * * * getWidth " +
                bufferedimage.getWidth());
    Tools.debug(" * * * Inside Capture 1 * * * getHeight " +
                bufferedimage.getHeight());

* * * Inside Capture 1 * * * getWidth 1280
 * * * Inside Capture 1 * * * getHeight 960

Thanks,
Arum

sarxos, 2017-09-09:
It will allow you to capture only these resolutions which are supported by your webcam. If you try to capture higher resolution then underlying framework will automatically switch to the closes supported one.

You see QXGA because you deliberately add it to the list of supported resolutions (custom view sizes) regardless the fact that your camera does not support it:

```java
Dimension[] nonStandardResolutions = new Dimension[] {
  WebcamResolution.PAL.getSize(),
  WebcamResolution.QXGA.getSize(),
  new Dimension(2000, 1000),
  new Dimension(1000, 500),
};
webcam.setCustomViewSizes(nonStandardResolutions); webcam.setViewSize(WebcamResolution.QXGA.getSize());
```

It doesn't matter if you set it when your camera is unable to handle it.

There are some limitations in the underlying framework that may render it impossible:
* No support for USB 3.0, or
* Non UVC-compliant camera.

arumbala72, 2017-09-09:
Thank you Bartosz ,is there any future plan for USB 3.0 support ? because now a day all the laptops and new computers are coming with USB 3.0

sarxos, 2017-09-09:
This depends on the framework used. As you may know it's not the Webcam Capture API that interacts with webcam hardware (through USB), but underlying framework which is used by a capture driver ([here](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers) is a list of available capture drivers).

Currently supported frameworks are:

* [VLC](https://www.videolan.org/vlc/) via [VLCj](https://github.com/caprica/vlcj)
* [V4L](https://en.wikipedia.org/wiki/Video4Linux) via [V4L](https://code.google.com/archive/p/v4l4j/)
* [VideoInput](https://github.com/ofTheo/videoInput) via [OpenIMAJ](http://openimaj.org/)
* [OpenCV](http://opencv.org/) via [JavaCV](https://github.com/bytedeco/javacv) (experimental)
* [LTI-CIVIL](https://sourceforge.net/projects/lti-civil/)
* [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html)
* [GStreamer](https://gstreamer.freedesktop.org/) via [gstreamer-java](https://github.com/gstreamer-java/gstreamer-java) (experimental)
* [FSWebcam](http://www.sanslogic.co.uk/fswebcam/)

As soon as USB 3.0 support is implemented in these frameworks it should start working with Webcam Capture API, however I guess the closest ones to introduce stable USB 3.0 support are OpenCV and GStreamer which both are used by experimental capture drivers.

arumbala72, 2017-09-09:
Thank you so much Bartosz,you are very helpful.I appreciate all your work.