sarxos/webcam-capture #424: try build webcam qr code in different mode

SPRL08, 2015-12-22:
how to build webcam qr code if i want the result like this
[image]

thanks before

sarxos, 2015-12-23:
@SPRL08,

You may start looking at the source code available in this example:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-qrcode

I'm building it using Maven, but you can get rid of it and use anything what you need. Medium Java experience is required to get started.

Java source code:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-qrcode/src/main/java/com/github/sarxos/webcam/WebcamQRCodeExample.java

sarxos, 2015-12-23:
@SPRL08,

Webcam panel and QR code extraction are covered in the example you. As for the database entries, there are many database frameworks available in the Java world. Some are pretty easy to start with and some are located on a very high abstraction layer.

If your Java experience is on a medium level, I would suggest to start with JDBC. The very fundamental database Java layer to access database. In JDBC you are using the same SQL syntax as your database is using.

https://docs.oracle.com/javase/tutorial/jdbc/basics/
http://www.mkyong.com/jdbc/how-to-connect-to-mysql-with-jdbc-driver-java/

If you are more experienced Java developer and do not need high database relations abstraction, you may take a look at JDBI. In JDBI you are using the same SQL syntax your database is using.

http://jdbi.org/

And finally, if you are a very experienced developer with a strong Java skills and you need many relations in your database model, then I would suggest to start with Hibernate. In Hibernate you are using abstract HQL language or the SQL syntax your database is using (by invoking native SQL queries).

http://www.mkyong.com/tutorials/hibernate-tutorials/