sarxos/webcam-capture #425: Unable to download JAR file

TorukoFenix, 2015-12-23:
Hi, I am trying to implement a QR Code Webcam scanner. However, I could not download the JAR file for the webcam capture from the link provided as I am redirected to a 404 page. Is it possible that the link is updated? Thank you.

sarxos, 2015-12-23:
To what link? All JARs are available from [Maven Central](http://search.maven.org).

sarxos, 2015-12-23:
You can also download ZIP. Check main page, section Download.

TorukoFenix, 2015-12-23:
Hi, it's the link that you have provided in the description of your GitHub project. I have already found the JAR file on Maven Central though, thanks a lot. I have referenced your webcam-capture JAR to my project already, including the ZXing JAR as well. Referring to your QR Code example, is there a reason why the private Webcam and private WebcamPanel declaration contains an error even after the referencing? I'm sorry if my questions seemed a little silly, I'm quite new to programming Java with the usage of an external library, and I'm kinda pressed for time. Thank you.

sarxos, 2015-12-23:
@TorukoFenix,

I found the broken link. It points to snapshot (latest build created from HEAD). It's stored in Sonatype repo, but they keep such builds only for some time (several months I think). From the quick inspection of the repo I see they removed old files. Just before a minute I deployed the newest (0.3.12) snapshot and fixed the link s it points to the correct version (it was pointing to 0.3.11).

In regards to your question:

> why the private Webcam and private WebcamPanel declaration contains an error even after the referencing

I do not understand what do you mean by _contains an error_. Can you please explain?

TorukoFenix, 2015-12-24:
There was the typical red line under the Webcam and WebcamPanel. When I mouse over to the error a "cannot find symbol" was shown. I'm currently using NetBeans by the way. Is it related to the way I name the class? I have named it to WebcamTester but I have already included the extends and implements. One thing I noticed as well was import com.google.zxing.client.j2se.BufferedImageLuminanceSource; had an error where they mentioned "package does not exist". However I could find that file inside the ZXing folder, although I'm using Core.JAR. Is it because of the way I import the library?

sarxos, 2015-12-24:
This is because you haven't include required JARs into the classpath.
Please google "how to add JAR to classpath in Netbeans".

TorukoFenix, 2015-12-24:
Yea, I have done that already, I've included core-2.2.jar for ZXing and webcam-capture-0.3.11.jar for your project. Is it somewhat related to the version of the JAR files?

TorukoFenix, 2015-12-25:
Hi, I have fixed the problem after slowly importing the usage of your class files. Thank you so much!

sarxos, 2015-12-25:
Cool! Marry Christmas and a Happy New Year! Take care :)

closed 2015-12-25 by sarxos

TorukoFenix, 2015-12-25:
Merry Christmas to you too, and a great new year!