sarxos/webcam-capture #426: Performance question

shayaantx, 2016-01-03:
Hi,

Thanks for your work on this library, it has made custom development with web cam + java really nice/seamless.

I've used api 3.1 across 3 different machines (two dell precision laptops with similair hardware and one beastly gaming desktop).  My weakest machine out of all 3 yields 14fps or higher using the web cam panel. My more powerful laptop always yields 5-6 fps (with spurts of 10-15fps). My gaming pc always yields 5fps.

I will say, the weakest machine shows the worst quality out of all 3, unless I shine a light behind webcam, when I do this, the fps still holds at around 14fps. The other two machines render pretty clear images. Been using 640 x 480 for resolution size, but this doesn't seem to be the red hearing. Decreasing to 320 x 200, does nothing to improve fps, other than on my weakest machine (can get up to 20fps when I start doing this).

I've went through the process of verifying other processes are not taking extra cpu/memory and I've gone through just having my code + web cam run alone on all 3 machines.

I've profiled the code using yourkit and saw the following. In a 11-12 second profile capture of the java code showing the web cam panel, SunGraphics2d#draw image was called 65 times with a total time of 11, 792 ms. Which makes sense if you consider I'm getting 5-6fps and each draw image call corresponds to a frame being drawn (so 6frames x 11 seconds = 66 calls, over 11,792 ms)

- By taking a trace of java sun2d rendering code, I can see its using direct x (so these render calls should be occurring on the gpu). Both my laptops are using direct x 11 and my gaming pc is using direct x 10.

What can I do to troubleshoot more? 

Thanks again

sarxos, 2016-01-03:
Hi @shayaantx,

If you are not using low power device (like Rasberry Pi), but just a regular, modern PC, the FPS should be limited only by the light conditions. The camera hardware adjust exposure time automatically, so when the light is dark, the exposure time is higher, thus making FPS lower. When light conditions are very good, the exposure time is small, and thus FPS is higher.

In most cases, when light conditions are good, the effective FPS is around 29-30 FPS (which is max FPS for most of the cameras).

Please tell me - how did you measure FPS? Is this a number that is displayed in the ```WebcamPanel``` corner? If so, then it's an effective capture performance (imaging, copying bytes from buffer and converting to ```BufferedImage```), without a rendering time, so drawing is not included here (capturing and drawing are in different threads).

You mentioned that you are using API 3.1, or did you mean 0.3.1? If so, please make sure to upgrade to the newest one, because there was a bug in a ```WebcamPanel``` rendering code which affected drawing performance.

shayaantx, 2016-01-03:
Thanks for responding, I upgraded to 3.11 and the issue persists. I'm using the code from (https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamPanelExample.java) so my code doesn't become the bottleneck.

I assumed that fps in the bottom corner represented the rendering (i now see your todo in webcam panel to rename it ;)). Regardless tho, capturing (not rendering) 5 frames a second isn't ideal, since actual rendering is limited by the capture frequency (if I'm understanding your comments correctly).

I've changed the lighting conditions with all 3 machines and not really had any success increasing fps or viewing a decrease. 

I might try a better webcam and see the results.

Thanks

shayaantx, 2016-01-03:
I played a little more with it and skype actually has an option to modify exposure for the web cam, Modifying this value improves the fps a lot.

Does your library expose any api for modifying parameters like this?

sarxos, 2016-01-04:
@shayaantx, nope, the Webcam Capture API is very simple and allows one to capture images, but does not give control over the advanced webcam parameters (like focus, exposure time, white balance). This is because I wanted to have common, unified API, regardless of the framework (or rather _capture driver_) used underneath (VLCj, OpenIMAJ, JMF, GStreamer, OpenCV, etc). Some frameworks gives access to these advanced parameters (e.g. JMF, OpenCV), but the other ones has no API for this, so adding such abstraction would be a little bit problematic.

sarxos, 2016-01-04:
@shayaantx, in regards to the webcam - I know that some have a possibility to fix exposure time to some predefined value via control panel. If you are using such feature please make sure you have fixed exposure time disabled.

closed 2016-04-19 by sarxos