sarxos/webcam-capture #427: motion detection and getting image on the same Webcam

creepid, 2016-01-04:
Hi,
i'm trying to use one camera from one classloader as motion detector and at the same time i would like to see image from this camera in another application (in fact from another classloader). 
I've encountered this reasonable exception:
com.github.sarxos.webcam.WebcamLockException: Webcam USB-2845>CAB@>9AB2> 0 has already been locked
Maybe exists some trick to get round this restriction?

Very thanks

sarxos, 2016-01-05:
Hi @creepid, 

Yes, there is indeed such option, but you have to be aware of the possible downsides. You can disable lock mechanism, but since webcam state is being kept in a Java code only, this may cause other issues, which are:
* On Linux, error from ```libv4l2``` saying that webcam cannot be open,
* On Windows, capture graph issue which may lead to the webcam subsystem failure (so you will have to restart PC to get it working again) and random application crashes (segmentation fault),
* Not tested on MacOS.

If you are in the same JVM instance, but have two separate classloaders, you should be easily able to exchange ```Webcam``` reference and use it from these two classloaders. However, if you are in a two distinct and separated JVMs, there is no way to use the same webcam, because you cannot exchange reference and cannot open it twice, which should be avoid due to the reasons I listed above.

What I would suggest (if you are in two separated JVMs) is to convert ```BufferedImage``` to ```byte[]``` and share it with second process via ```Socket```, RMI, or e.g. memory mapped file:

http://stackoverflow.com/questions/25396664/shared-memory-between-two-jvms

If you still want to release the lock, then:

```java
Webcam w = Webcam.getDefault();
WebcamLock lock = w.getLock();
if (lock.isLocked()) {
    lock.unlock();
}
```

Or disable it (so it won't lock):

```java
Webcam w = Webcam.getDefault();
WebcamLock lock = w.getLock();
lock.disable();
```

But I **strongly** suggest to keep it as it is and don't mess with locks. The locking mechanism has been designed to prevent many problems (especially on Windows, because it allows two processes to use the same capture graph) which caused memory segmentation faults and complete webcam subsystem failure.

creepid, 2016-01-05:
@sarxos, thanks for so detailed response, i have two separate classloaders in the same JVM instance on Windows, where after lock disabling this exception occured (i mean the second classloader, in the first classloader i'm using ```Webcam``` reference for motion detection at this moment with disabled lock for the ```Webcam```).
Here is the exception:
```JAVA
com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
......
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:335)
```
although i tried to check whether the camera is open by calling ```webcam.isOpen() ``` (second classloader) it always returns ```"false"```. Now i'm trying to exchange reference between two separate classloaders

sarxos, 2016-01-07:
@creepid, are you using ```Webcam.getDefault()``` in both classloaders, or only in one, and then pass ```Webcam```` reference to the second classloader? In first case the static initialization cause obtained ```Webcam``` reference to be a different object pointing different native memory. In the second case you should be able to use both references normally.

sarxos, 2016-01-07:
In first classloader:

```java
Webcam x = Webcam.getDefault();
x.isOpen(); // false
x.open();
x.getImage(); // fetch frame from webcam buffer
```

Doing  the same in 2nd classloader will cause errors, so you should do something like this:

```java
Webcam y = x;
// don't open, it's already open in 1st classloader
y.isOpen(); // true
y.getImage(); 
```

sarxos, 2016-01-07:
I'm just curious. Are you maybe using OSGi? I'm asking because it's pretty rare for someone to use distinct classloaders in non-OSGi applications.

creepid, 2016-01-07:
Pretty clear, thanks. 
No, i'm not using ```OSGi```, but just trying to expose cameras streams to a web page and at the same time use some of webcams as motion detector devices on the back-end side. 
I've used ```webcam-capture``` project and it's examples as camera-management part of my experimental home-made project https://bitbucket.org/mikhail_complete/smarthome/src and found it very useful :)

sarxos, 2016-01-09:
Hi @creepid, you have very nice project done there ;) Is it possible to run it on Linux? I played a little bit with onewire in the past but I had to have Maxim drivers installed on my Windows to have their OW library working with USB OW controller (DS2490 if I remember correctly). Unfortunately, without these drivers installed, their library didn't work on Linux then, in 2010, but maybe something changed nowadays?

creepid, 2016-01-09:
actually, you may use RXTX lib to get it work on Linux. But there is no available owapi.jar in any public repo and it is no longer developing :( (last version - 1.10). So, i've tried to store last snapshot [here](https://github.com/creepid/owapi), and if you are only interested in onewire-part of app you may use [this one](https://github.com/onewire/onewire). Actually, my project is just a fork of it and has a lot of architectural issues.
p.s. your https://github.com/sarxos/1-wire-network project is right what i've suspected but have no idea how to do it right :) i'll learn it in details

sarxos, 2016-01-11:
Yes, I thought about RXTX, but I remember USB being much simpler with owapi. In regards to my project - I wrote it years ago, it has dozens architectural and design issues which I haven't resolve and I would treat rather as a proof-of-concept than a working system. The ideas behind were pretty nice, but the code... well, today it is like this:

http://devopsreactions.tumblr.com/post/66268086944/looking-at-what-you-wrote-yesterday