sarxos/webcam-capture #429: Potential causes of "Timeout when requesting image!" in WebcamDefaultDevice.updateFrameBuffer?

phrack, 2016-01-08:
com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice in updateFrameBuffer at line 479 (application)
com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice in run at line 500 (application)

Any ideas what would cause this? We have automatic error reporting set up for our fairly popular application and we get this report hundreds of thousands of times a week. Unfortunately, we can't really tell you what cameras are being used because this exception is essentially swallowed as far as users of webcam-capture are concerned (i.e. we wouldn't even know it is happening without the error reporting).

sarxos, 2016-01-08:
Hi @phrack, 

This happens when next frame call return -1 ([here](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L479)) which is caused by video data image buffer not ready to be read ([here](https://github.com/openimaj/openimaj/blob/master/hardware/core-video-capture/src-native/windows/OpenIMAJGrabber/OpenIMAJGrabber.cpp#L187) and [here](https://github.com/ofTheo/videoInput/blob/22356ca0c4808ae4c6e5ed9546574e8a7bf347be/videoInputSrcAndDemos/libs/videoInput/videoInput.cpp#L1046)), so essentially when there is no new image from camera in 5 seconds. The webcam has been already open, but fetch new image request is waiting for to be completed and finally timeouts.

I suppose this may be somehow connected with the USB device being powered down when OS disables USB port to lower power consumption, but this is just the idea, not empirically proven statement. For more detail on how to enable/disable this USB feature you may take a look at this and test it yourself:

http://helpdeskgeek.com/windows-xp-tips/prevent-windows-from-powering-off-usb-device/

Just for your information - this error is not a critical exception. It's logged as error since it's pretty unexpected, but the application works normally (there is no ```throw``` statement in the code for this). The only issue is a video being frozen.

You can try enabling debug, but you would have to report whole console output, not only lines with ERROR. Adding ```-Dwebcam.debug=true``` into the ```java``` command and setting ```OPENIMAJ_GRABBER_VERBOSE=true``` environment variable could provide some more details from the depths of native code (more details can be found [here](https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs)).

maxxx580, 2017-08-07:
Hello @sarxos 

I am having the same problem where the program timed out. Could this be a hardware dependent problem? 

Thanks

Eric

zeyuanzhang, 2018-10-15:
I'm having the same problem with Mac webcam FaceTime HD Camera (Built-in).

oori, 2020-12-25:
@sarxos Regarding your comment:

> Just for your information - this error is not a critical exception. It's logged as error since it's pretty unexpected, but the application works normally (there is no throw statement in the code for this). The only issue is a video being frozen.

I'd like to handle this inside my application, by restarting the camera.   if I do this manually (restart my app) it always works.   as there's no exception throw, how could I catch this?    (application keeps running normally, frozen video)

It's a hard issue to solve due to inconsistency.   Test app works for hours long and never crashes, but with my app, which does a lot of visual processing, has irregular frame rate and high CPU load - it happens every half an hour.

Thanks.

diveair, 2021-01-11:
Is there any way to catch the frozen video as some sort of event? May need to modify the WebcamDefaultDevice updateFrameBuffer() method to add the event there. 

I have been unable to find any other solution to it.