sarxos/webcam-capture #430: Java crashes when running WebcamViewerExample

coolosity, 2016-01-08:
So whenever I try to run WebcamViewerExample, windows comes up with Java(TM) Platform SE binary has stopped working. Once instead of this message it just showed a black screen. I've been looking through other issues and haven't found anything similar to mine, so sorry if I've missed one and am asking something again. I am running Windows 7 x64 with java 8u66

sarxos, 2016-01-09:
Hi @coolosity, yes, some webcams may cause problems (especially on Windows). Do you have some dedicated webcam drivers installed for it or is it PnP? Also, does it create some crash log on exit?

coolosity, 2016-01-09:
I checked and the currently installed driver is up to date (6.1.7601.18208). The webcam also works with other non-java programs. The hardware ids are
USB\VID_064E&PID_8128&REV_1<13&MI_00
USB\VID_064E&PID_8128&MI_00

I haven't been able to find a java crash log, but here is the windows log:
http://pastebin.com/BBfgGr7u

sarxos, 2016-01-11:
@coolosity, you may try using different capture driver, e.g. you could install [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html) and use [webcam-capture-driver-jmf](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-jmf) instead of a default one.

sarxos, 2016-02-10:
@coolosity, is it possible for you to make the same run but using JDK instead of JRE? By using JDK we will have more detailed stack trace for this problem (now we have no stack trace at all).