sarxos/webcam-capture #431: OpenIMAJGrabber DLL identified as cause of crash

bxnkole, 2016-01-19:
Hi,

I'm using the default OpenIMAJGrabber driver with the webcam-capture API. However, after some captures, my application crashes with the prompt "Java Binary Platform Has Stopped Working".

Of course, a crash dump log file was generated (pasted below) and it identified OpenIMAJGrabber dll as the offending frame. I've deployed my application to over 1000 Windows (x64 and x86) system as standalone applications and it keeps crashing after a few captures. 

From the crash logs, it appears it has something to do with setting timeout, i'm not sure. The application's logic has to do with biometric registration so captures of several persons are taken one after the other.

API Version: 0.3.10
Java Version Deployed: 1.7 update 79

I will be wiiling to divulge more information. Thank you. 

Crash Report

 A fatal error has been detected by the Java Runtime Environment:

  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007fff8b47a3d0, pid=16572, id=16984

 JRE version: 7.0_21-b11
 Java VM: Java HotSpot(TM) 64-Bit Server VM (23.21-b01 mixed mode windows-amd64 compressed ops)
 Problematic frame:
 C  [OpenIMAJGrabber.dll+0xa3d0]  OpenIMAJGrabber::setTimeout+0x83b0

 Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

 If you would like to submit a bug report, please visit:
   http://bugreport.sun.com/bugreport/crash.jsp
 The crash happened outside the Java Virtual Machine in native code.
 See problematic frame for where to report the bug.

---------------  T H R E A D  ---------------

Current thread (0x000000000c59d800):  JavaThread "atomic-processor-1" daemon [_thread_in_native, id=16984, stack(0x000000000ea30000,0x000000000eb30000)]

siginfo: ExceptionCode=0xc0000005, writing address 0x0000000014e4beaa

Registers:
RAX=0x0000000000000085, RBX=0x0000000000000000, RCX=0x0000000014e4beaa, RDX=0x0000000014d01090
RSP=0x000000000eb2f090, RBP=0x000000000eb2f1f0, RSI=0x000000000be36a20, RDI=0x000000000be375d0
R8 =0x0000000014de20a0, R9 =0x0000000000000280, R10=0x0000000000000000, R11=0x0000000000000000
R12=0x000000000be36990, R13=0x000000000eb2f2e0, R14=0x000000000eb2f2e0, R15=0x000000000eb2f3c8
RIP=0x00007fff8b47a3d0, EFLAGS=0x0000000000010283

Top of Stack: (sp=0x000000000eb2f090)
0x000000000eb2f090:   0000000000000000 0000000000000000
0x000000000eb2f0a0:   0000000000000000 0000000000000000
0x000000000eb2f0b0:   000e100000000780 0000000000000000
0x000000000eb2f0c0:   000e088000000000 00069e0a0000022e
0x000000000eb2f0d0:   000000000eb2f2e0 000000000be375d0
0x000000000eb2f0e0:   000000000be36a20 00007fff8b4785ba
0x000000000eb2f0f0:   000000000ec3e580 0000000014d7881a
0x000000000eb2f100:   0000000014e4beaa 0000000000000280
0x000000000eb2f110:   0000512d000001e0 00000000a2740d01
0x000000000eb2f120:   0000512d06d5c001 000000000eb2f2d0
0x000000000eb2f130:   000000000eb2f300 0000000014de20a0
0x000000000eb2f140:   000001e000000280 00007fff00000000
0x000000000eb2f150:   0000000014d01090 0000000000000000
0x000000000eb2f160:   0000000000000000 0000000075a512d7
0x000000000eb2f170:   0000512d06d50100 000000000c59d9d8
0x000000000eb2f180:   000000000eb2f2e0 00007fff8b471fef 

Instructions: (pc=0x00007fff8b47a3d0)
0x00007fff8b47a3b0:   d1 e0 48 63 c8 48 8b 44 24 68 48 2b c1 48 89 44
0x00007fff8b47a3c0:   24 68 48 8b 4c 24 70 48 8b 44 24 68 0f b6 40 02
0x00007fff8b47a3d0:   88 01 48 8b 44 24 70 48 83 c0 01 48 89 44 24 70
0x00007fff8b47a3e0:   48 8b 4c 24 70 48 8b 44 24 68 0f b6 40 01 88 01 

Register to memory mapping:

[... 335 lines cut ...]

VM Arguments:
jvm_args: -Dfile.encoding=UTF-8 
java_command: com.sf.biosmart.BioSmartController
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\Java\jdk1.7.0_21
PATH=C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Java\jdk1.7.0_21\jdk1.7.0_45\bin;C:\Program Files\TortoiseSVN\bin;C:\apache-ant-1.9.6\bin;M2_HOME\bin;C:\Program Files\Java\jdk1.7.0_21\bin;C:\apache-maven-3.3.3\bin;C:\JCodes\biosmart\native;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\Skype\Phone\;C:\MinGW\msys\1.0\bin
USERNAME=Kole
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 8 , 64 bit Build 9200 

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 58 stepping 9, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, ht, tsc, tscinvbit, tscinv

Memory: 4k page, physical 5857140k(1016268k free), swap 12743084k(2183700k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (23.21-b01) for windows-amd64 JRE (1.7.0_21-b11), built on Apr  4 2013 08:11:28 by "java_re" with unknown MS VC++:1600

time: Sun Jan 17 20:12:50 2016
elapsed time: 60 seconds

sarxos, 2016-02-10:
Hi @bxnkole,

Thank you for the report. This should be now fixed.

bxnkole, 2016-05-10:
Thanks @sarxos, will there be a release for this fix?

sarxos, 2016-05-10:
Can you please try [snapshot](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.12-SNAPSHOT) and verify if change implemented in 2ccfaed fixed this issue?

bxnkole, 2016-05-10:
Ok I will. Thank you.

bxnkole, 2016-05-11:
@sarxos I just did a series of captures with the snapshot. The crash occurred about 20 minutes later. There was no crash dump this time but I enabled debug mode. See logs below

`2016-05-11 16:18:51,493 DEBUG ImageCaptureDialog:930 - *** Attempting to capture a passport.
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Created native object from pointer Pointer(peer = 0x68d280, targetType = com.github.sarxos.webcam.ds.buildin.natives.DeviceList, order = LITTLE_ENDIAN)
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.DeviceList.getNumDevices()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.DeviceList.getDevice(int)
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Created native object from pointer Pointer(peer = 0x68d760, targetType = com.github.sarxos.webcam.ds.buildin.natives.Device, order = LITTLE_ENDIAN)
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.DeviceList.getNumDevices()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method protected native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.Device.getName()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method protected native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.Device.getIdentifier()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getImage()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:51 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:18:52 PM org.bridj.BridJ log
[... cut ...]
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()
May 11, 2016 4:19:13 PM org.bridj.BridJ log
INFO: Calling method public native org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getImage()
May 11, 2016 4:19:13 PM org.bridj.BridJ log
INFO: Calling method public native int com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()`

Although I didn't notice anything out of the ordinary here.

Thank you for your support.

blacktorch, 2018-12-07:
This issue still persists on 64 bit Java builds , is there any fix for this yet. Thanks

wdp1990821, 2020-05-20:
This issue still persists on 64 bit Java builds , is there any fix for this yet. Thanks

commit 2ccfaed4 (2016-02-10): Set timeout once, after grabber is open, fixes #431