sarxos/webcam-capture #432: Possible memory leak when using nonStandardResolutions in JavaFX

bogey4, 2016-01-27:
I have adapted the CustomResolutionExample.java example to work with the webcam-capture-javafx example, and have encountered a potential memory leak. This was observed by profiling memory consumption in Netbeans 8.0.2 with java 1.8.60, and noticing that the Surviving Generations was constantly increasing. There was no such issue observed when the setCustomViewSize and setViewSize methods were not called.

To recreate: 

`    protected void initializeWebCam(final int webCamIndex) {

       Task<Void> webCamTask = new Task<Void>() {

         @Override
         protected Void call() throws Exception {

           if (webCam != null) {
             disposeWebCamCamera();
           }

           webCam = Webcam.getWebcams().get(webCamIndex);
           Dimension[] nonStandardResolutions = new Dimension[]{
             WebcamResolution.PAL.getSize(),
             WebcamResolution.HD720.getSize(),
             new Dimension(2000, 1000),
             new Dimension(1000, 500),};

           webCam.setCustomViewSizes(nonStandardResolutions);
           webCam.setViewSize(WebcamResolution.HD720.getSize());
 
          webCam.open();
           startWebCamStream();

           return null;
         }
       };

       Thread webCamThread = new Thread(webCamTask);
       webCamThread.setDaemon(true);
       webCamThread.start();

       bottomCameraControlPane.setDisable(false);
       btnCamreaStart.setDisable(true);
     }`
[image]

Please let me know if I can clarify further.

sarxos, 2016-02-10:
Hi @bogey4,

I did a little profiling and it seems that old gen is clogged by the ```BufferedImage``` instances which has not yet been converted into JavaFX ```Image``` because conversion between these two happens in JFX main rendering thread slowing it down and causing this issue to be multiplied by slowing conversion rate many times.

closed 2016-02-10 by sarxos

sarxos, 2016-02-10:
With this change memory consumption should be much smoother. Also rendering speed should perform faster on HD720p.

sarxos, 2016-02-10:
I noticed that many heavy objects are created in runtime by JavaFX. These are rasters and images which are being held by the JFX internals. Not sure how to get rid of them.

bogey4, 2016-02-24:
Sarxos, thanks for looking into this issue. I pulled the changeset referenced above, but still have the same issue, wherein the objects are surviving indefinitely (see attached profiler image). Is there any path forward to resolve this, or is the leak to remain? Again, thanks for your excellent work on this project. It's greatly appreciated! 

[image]

sarxos, 2016-02-25:
Hi @bogey4,

Can you find out what are the objects that are clogging your memory? I used [Eclipse Memory Analyzer (MAT)](http://www.eclipse.org/mat/) for this. First I'm doing heap dump from ```jvisualvm``` and then opening it from MAT. You may need to add ```-Xmx2048m``` in ```MemoryAnalyzer.ini``` if memory dump is very big.

I'm almost sure those memory clogs are created by JavaFX, so I think you can definitely get rid of this problem by not rendering live image preview from JavaFX, but use ```WebcamPanel``` instead (it's Swing object).

commit 158b65a7 (2016-02-10): Fix rendering memory leak in JavaFX example, fixes #432