sarxos/webcam-capture #434: java.lang.UnsatisfiedLinkError: no video in java.library.path

mjoycemilburn, 2016-02-07:
Hi Sarxos,

I'm trying to do a Maven build the Play Video from Camera example under WIndows 10 using the V4l4J library. I get the above error on  "w.setDriver(new V4l4jDriver());" Is this telling me I need to find a Natives library somewhere? If so, does such a thing even exist for Windows? I'm starting to wonder if perhaps V4L4J is just for LInux users

sarxos, 2016-02-08:
V4L4J stands from _Video For Linux For Java_ - this is pure Linux stuff, no Windows support.

sarxos, 2016-02-08:
On Windows you can try OpenIMAJ, GStreamer or OpenCV.

closed 2016-02-08 by sarxos

mjoycemilburn, 2016-02-08:
Thanks for that Sarxos - I'm still chasing the intermittent camera timeouts I'm getting with the default OpenImaj driver. My camera is a Polaris CCTV job and I connect it to my PC with a composite/USB adapter. This uses drivers from August and the camera works fine with the supplied CCTV utility. 

Webcam-capture struggles,however, and since I've got the latest version of the August drivers, playing around with the Capture Drivers is all I can think to do.

If I run out of options here can you recommend a useful low-cost IP or USB camera?