sarxos/webcam-capture #435: daemon thread status in streaming thread

teamlead-agbaje, 2016-02-09:
Hi,

First, awesome API.

Now, according to the example, the thread that handles the streaming for the webcam has a daemon status set to true.

However, I've noticed that sometimes, after shutting down my javafx application and starting it up, the daemon thread doesn't start up, and so my application doesn't pick up anything from the webcam.

When I take off the daemon status, everything works fine.

I just need to know, why are you using a daemon thread? Is it Ok if I don't? If I must, how do I get around this issue?

Thanks.

And thanks again for this - this is a truly wonderful API.

teamlead-agbaje, 2016-02-09:
Hi, I actually narrowed the problem down. Apparently, it's not the daemon thread, and changing it to a normal thread really doesn't solve the issue.

The real issue is this: grabbedImage = selWebCam.getImage();

Sometimes, selWebCam returns an image for grabbedImage. Other times, it doesn't. selWebCam doesn't get an image at all, and so nothing is shown.

How do I solve this problem, please?

sarxos, 2016-02-10:
Hoi @teamlead-agbaje, is this IP or USB camera?

teamlead-agbaje, 2016-02-10:
Hi, the camera comes built with my laptop. I use it to skype all the time.

sarxos, 2016-02-10:
Usually you get ```null``` image when camera has been closed in the middle of use. E.g. you fetch images in one thread, and in another thread you invoke ```close()```. This may cause ```null``` to be returned.

And in regards to this:

> why are you using a daemon thread? Is it Ok if I don't?

The deamon threads are threads which will die when main thread has completed. All non-daemon threads will remain alive and prevent application from being stopped. This is in regards to the classic application.

In regards to this JavaFX example I'm not sure when and why daemon threads need to be used because it's not me who wrote these examples. Both JavaFX examples has been prepared by generous people from community who are much more familiar with the JavaFX.

I propose to test your camera with plain old Java example - e.g. this one:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamViewerExample.java

If you notice the same issue as you described, then indeed, something wrong may be happening in the API, but if there is no issue, then I would suppose it's a JavaFX example somehow buggy.

teamlead-agbaje, 2016-02-10:
Hi sarxos,

Thanks so much. After you mentioned that the getImage method will return null if the web cam was closed in operation, I went back to scrutinise my codes. Turns out while I'm calling the method to start the thread for the camera, I have a change listener that's calling the same thread method, and so the if statement where selWebCam is checked for null is run twice, once when it is null and when it isn't, which leads to the web cam closing before its begun streaming.

I've resolved the issue now. Thanks so much!

closed 2016-02-14 by teamlead-agbaje