sarxos/webcam-capture #436: How to shutdown the camera?

Nyutkarsh, 2016-02-10:
hi, 

I have tried all the method for closing or shutting down the camera but nothing works. Will you please tell me how can close or shut down the camera which is being used.

I want to close it when I click the "X" button on JFrame.
And Same at the time of session time out and Logout.

Nyutkarsh, 2016-02-10:
@sarxos

sarxos, 2016-02-10:
@Nyutkarsh, simply do ```webcam.close()``` in your ```windowClosed(WindowEvent e)``` method that comes from ```WindowListener``` interface.

https://docs.oracle.com/javase/tutorial/uiswing/events/windowlistener.html

closed 2016-02-10 by sarxos

sarxos, 2016-02-10:
See this example:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamViewerExample.java

Line 108:

```java
	@Override
	public void windowClosed(WindowEvent e) {
		webcam.close();
	}
```