sarxos/webcam-capture #437: Webcam quality

teamlead-agbaje, 2016-02-14:
Hi,

I increased the size of the imageview (javafx) where the buffered grabbed image from the web cam will be displayed and despite that, the picture quality appears rather pixelated. Is there some code I can add to make the web cam quality better? It appears, the quality is only good when the imageview is small.

teamlead-agbaje, 2016-02-14:
Ok, I've managed to clean up the grabbed image by playing around with the web cam resolution setting, within dimensions.

However, I've come to notice that the images are coming up inverted. That is, when I raise my right hand, the image shows me raising my left hand haha! This has nothing to do with the resolution setting. Even without it, the image is still inverted. Is there a way to set the inversion?

sarxos, 2016-02-14:
Hi @teamlead-agbaje,

If you are using ```WebcamPanel``` to display image from webcam, then it's enough to call:

```java
panel.setMirrored(true);
```

But if you do not use ```WebcamPanel``` (like in the javafx example), then you have to add ```WebcamTransformer``` to your ```Webcam``` instance. I do not have ready to use example demonstrating how to mirror the image, but this is fairly simple and these examples should give you some better understanding what needs to be done:

[ImageTransformerRotationExample.java](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/ImageTransformerRotationExample.java)

[ImageTransformerExample.java](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/ImageTransformerExample.java)

Just FYI, transformation feature converts image before it's returned from ```webcam.getImage()``` method.

closed 2016-02-14 by sarxos