sarxos/webcam-capture #439: Driver for MAC 64bits

diogo10, 2016-02-15:
Hi there,

I've been trying run with the drives from the main pages, but I didn't get to work. I'd like to test using the mac default webcam. How can I do it?

sarxos, 2016-02-16:
Which drivers?

diogo10, 2016-02-16:
I tried JMF and FFmpeg-cli but I guess that I need to install first the drives and then run my code. Is that correct?

sarxos, 2016-02-16:
Yes. It's correct. There drivers requires software which needs to be installed first. Try running your example without a driver (Webcam Capture it will use refined OpenIMAJ in this case).

sarxos, 2016-02-26:
@diogo10, I'm closing this one. Please let me know if you have any further issues in regards to this.

closed 2016-02-26 by sarxos

diogo10, 2016-02-26:
Well, didn't work at all. I've tried OpenIMAJ, FFmepg, OpenCV. I will make up the app in Swift, but the windows version I will try to use the library.

sarxos, 2016-02-26:
Ok, thank you for the info. I should find some newer Mac to perform tests and find out what is wrong. The last time I did it I had 6 years old Mac borrowed from my friend.

mjm-cycronix, 2017-04-21:
OpenIMAJ works on my 2012 iMac, OS Sierra 10.12.4.  It does spin the CPU 100% no matter how slowly I call getImage(), however.  Suspect still the old bug, reported:
https://github.com/sarxos/webcam-capture/pull/257