sarxos/webcam-capture #442: Webcam Driver missing

Sai-Thippasani, 2016-02-25:
The link provided to webcam-capture-driver-v4l4j-0.3.11-20150713.101304-10.jar in [this ticket](https://github.com/sarxos/webcam-capture/issues/375) is broken. Only version 3.10 is available

Can you please fix it?

Thank you

sarxos, 2016-02-26:
Hi @Sai-Thippasani,

Thank you for the report.

You can find the latest and stable JAR in Maven Central:

http://search.maven.org/#artifactdetails|com.github.sarxos|webcam-capture-driver-v4l4j|0.3.11|jar

Don't forget to add this JAR as well. It's a transitive dependency.

http://search.maven.org/#artifactdetails|com.github.sarxos|v4l4j|0.9.1-r507|jar

Take care!

sarxos, 2016-02-26:
I've open separate #444 to provide downloadable ZIP bundle for V4L4J and I'm closing this one. Please let me know if you have any further issues.

closed 2016-02-26 by sarxos