sarxos/webcam-capture #446: Webcam-capture with QR code

renhibari, 2016-02-27:
Hi sarxos! I have an error that says: 
package com.google.zxing.client.j2se does not exist
import com.google.zxing.client.j2se.BufferedImageLuminanceSource;

How can I resolve this issue? I know for myself that I've downloaded all the jar files I need to run this program. Can you give me exact .jar to download because maybe I was wrong after all. Thanks! 

And my other 6 errors are:
private Webcam webcam = null;
private WebcamPanel panel = null;
Dimension size = WebcamResolution.QVGA.getSize();
webcam = Webcam.getWebcams().get(0);
panel = new WebcamPanel(webcam);
LuminanceSource source = new BufferedImageLuminanceSource(image);

Thank you! And I chose this as my project. I just wanted to know about your program! :) My plan is by using this qrcode scanner, for example, have a unique that will need to scan by this program then it will serve as their attendance ^_^

[image]

sarxos, 2016-02-27:
I'm almost 100% sure that you haven't import zxing JAR or import incorrect version. What JARs have you included into your application's classpath?

sarxos, 2016-02-27:
I see your issues:

1. You are using wrong version of zxing core. This example [has been build on top of 2.1](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-qrcode/pom.xml#L26), and you included 3.11 instead.
2. You are missing zxing javase JAR (link below).

http://search.maven.org/#artifactdetails|com.google.zxing|javase|2.1|jar
http://search.maven.org/#artifactdetails|com.google.zxing|core|2.1|jar

I suggest to start using Maven to build your project. Maven manage JAR dependencies for you so issues like this does not happen.

closed 2016-02-27 by sarxos

sarxos, 2016-02-27:
You are also missing JARs which are required by Webcam Capture API to work (for BridJ and SLF4J). These can be found in ZIP file in ```libs``` directory. The ZIP can be found on project main page in section Download.

https://github.com/sarxos/webcam-capture/releases/download/webcam-capture-parent-0.3.10/webcam-capture-0.3.10-dist.zip

If you are Java beginner than I propose you to use IntelliJ, Eclipse or Netbeans IDE to write and compile your Java code. Especially IntelliJ has very nice support and automatically discover what you are missing and what may be wrong in your code.

Take care!

renhibari, 2016-02-28:
I've done what you said earlier and the only thing left is on the image:

[image]

Still got problems on the webcam part. I'm so sorry for being dumb >_<. As you can see in the image, I've already added the .jar files that you gave to me. Or... I was wrong again.

I'm so sorry again for disturbing you.

renhibari, 2016-02-28:
Ooops! I got the issue resolved! Thank yooooou! But is there any problem with these:

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

But the program works well as expected.

sarxos, 2016-02-29:
Hi @renhibari, I'm glad you managed to make it work. In regards to SLF4J warning - this is not error. It just tells you that you haven't add any logger binding to the classpath. In case you need logging to the console just use [Logback](http://logback.qos.ch/) (it is binding for [SLF4J](http://www.slf4j.org/)).

renhibari, 2016-03-02:
Ohh. Okay! Thanks again! :) God Bless! :+1: