sarxos/webcam-capture #447: java.lang.NoClassDefFoundError: ch/qos/logback/core/Context

MarcJose, 2016-03-06:
Hello,
I am using Maven to ingreate this library but when I build with _`mvn clean compile assembly:single`_ and run the generated jar-file I get:
```
java.lang.NoClassDefFoundError: ch/qos/logback/core/Context
        at com.cam.App.main(App.java:47)
Caused by: java.lang.ClassNotFoundException: ch.qos.logback.core.Context
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 1 more
```

When I run it with eclipse everything seems to be fine but all I want is a working runnable jar-file.
Hope someone can help.

MarcJose, 2016-03-06:
The example pom.xml was not complete...

sarxos, 2016-03-07:
Hi @MarcJose,

Thank you for the report.

I never tested how ```WebcamLogConfigurator``` behaves with ```assembly:single``` and it seems being broken. But worry not. This is just utility I wrote to have some more logging options, but in reality, if you are on Logback, you can keep its features by appropriate configuration in your application's executable command line:

```plain
java -Dlogback.configurationFile=/path/to/logback.xml com.cam.App
```

Or in runtime (in the very static main, before LOG is initialized):

```java
public class App {
  static {
    System.setProperty("logback.configurationFile", "/path/to/logback.xml");
  }
}
```

And don't forget to configure your Logback configuration to scan for the changes in loggers settings:

```xml
<configuration scan="true" scanPeriod="5 seconds" >
  ...
</configuration>
```

closed 2016-03-26 by MarcJose