sarxos/webcam-capture #448: WebcamDefaultDriver capture driver will be used

jeftanathan, 2016-03-07:
i'm wanna try this code
`import com.github.sarxos.webcam.Webcam;

public class Testcamera {

    /**
     * @param args the command line arguments
     */
    public static void main(String[] args) {
        Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
		} else {
			System.out.println("No webcam detected");
		}
    }`

output run : 
[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
Webcam: Integrated Webcam 0

that is output from IDE netbeans java
i'm first to use this lib, some body can help me??

lib :
- webcam-capture-0.3.10.jar
- slf4j-api-1.7.18.jar
- slf4j-simple-1.7.18.jar
- bridj-0.6.2.jar

- windows 8
- IDE NetBeans 8.0.2
- JDK 1.8

sarxos, 2016-03-07:
Hi @jeftanathan,

Is there any problem with this code?

The INFO statement just informs you what driver implementation will be used. It could use different capture driver (non-default, such as GStreamer, OpenCV), but you haven't configure any.

It detects your webcam as it was intended and it's name is "Integrated Webcam 0":

```
Webcam: Integrated Webcam 0
```

jeftanathan, 2016-03-07:
i'm sorry.. thanks for your respond.. this is first time i'm used this library.. I became a little confused.. but, i've been able to use it.. and sorry for my bad english..

sarxos, 2016-03-07:
Not a problem at all :) Just open new ticket if you have any other issues. I will close this one. Good luck!

closed 2016-03-07 by sarxos

jeftanathan, 2016-03-08:
thank you very much..