sarxos/webcam-capture #450: Exception in thread "JavaFX Application Thread" java.lang.RuntimeException: java.lang.reflect.InvocationTargetException

alexsrs, 2016-03-17:
I'm having trouble using your library in my project . I could do it compile but to run
 it runs more could not put the picture in my registration scene.
My code is here https://github.com/alexsrs/rfid-eventos/

I need help, can you help me ?

sarxos, 2016-04-19:
Hi @alexsrs,

Can you provide full stack trace? Not only exception class name?

hellmer1234, 2016-05-29:
Hi @sarxos ,

I have the same issue. The full stack trace provided is following : 
Exception in thread "Thread-8" java.lang.IllegalStateException: Not on FX application thread; currentThread = Thread-8
	at com.sun.javafx.tk.Toolkit.checkFxUserThread(Toolkit.java:204)
	at com.sun.javafx.tk.quantum.QuantumToolkit.checkFxUserThread(QuantumToolkit.java:364)
	at javafx.scene.Scene.addToDirtyList(Scene.java:485)
	at javafx.scene.Node.addToSceneDirtyList(Node.java:424)
	at javafx.scene.Node.impl_markDirty(Node.java:415)
	at javafx.scene.image.ImageView.access$600(ImageView.java:143)
	at javafx.scene.image.ImageView$3.invalidated(ImageView.java:288)
	at javafx.beans.WeakInvalidationListener.invalidated(WeakInvalidationListener.java:83)
	at com.sun.javafx.binding.ExpressionHelper$SingleInvalidation.fireValueChangedEvent(ExpressionHelper.java:137)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.ReadOnlyObjectPropertyBase.fireValueChangedEvent(ReadOnlyObjectPropertyBase.java:74)
	at javafx.scene.image.Image$ObjectPropertyImpl.fireValueChangedEvent(Image.java:568)
	at javafx.scene.image.Image.pixelsDirty(Image.java:542)
	at javafx.scene.image.WritableImage$2.setPixels(WritableImage.java:221)
	at javafx.embed.swing.SwingFXUtils.toFXImage(SwingFXUtils.java:129)
	at com.lmr.photobooth.WebCamAppLauncher$4.call(WebCamAppLauncher.java:263)
	at com.lmr.photobooth.WebCamAppLauncher$4.call(WebCamAppLauncher.java:1)
	at javafx.concurrent.Task$TaskCallable.call(Task.java:1409)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.lang.Thread.run(Thread.java:745)

Full source available 
[WebCamAppLauncher.java.zip](https://github.com/sarxos/webcam-capture/files/288297/WebCamAppLauncher.java.zip)

Thanks for your advices