sarxos/webcam-capture #452: How can record video in the javafx using sorex

vascubrian, 2016-04-23:


sarxos, 2016-04-24:
Hi @vascubrian, sorry to disappoint you, but I have no idea what the _sorex_ is :(

vascubrian, 2016-04-24:
Using xuggler, im recording video but there is no volume in java

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-04-24:
Is my final year project

sarxos, 2016-04-25:
The Webcam Capture API does not given any ready to use method for thois because it captures video frames only, not audio. However, if you want to encode video+audio using Xuggler, then you should just use ```writer.encodeVide(..)``` to encode video into your movie clip, and ```writer.encodeAudio(..)``` to encode audio. There are number of examples on the internet (simple google search is enough to find these):

http://stackoverflow.com/questions/11518164/how-i-can-use-xuggler-to-capture-audio-and-video

https://github.com/artclarke/xuggle-xuggler/blob/master/src/com/xuggle/mediatool/demos/GenerateAudioAndVideo.java

http://stackoverflow.com/questions/11647449/how-to-synchronise-audio-and-video-using-xuggler

But please note you have to deal with audio on your own since Webcam Capture API does not expose audio!

vascubrian, 2016-04-25:
So i need to record audio and video at same time ,then i combine two files
audio and video to get one file.

In one execution

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-04-25:
package startCamera;
import org.opencv.core.MatOfByte;
import org.opencv.imgcodecs.Imgcodecs;
import org.opencv.videoio.*;
import javafx.event.EventHandler;
import javafx.scene.control.Label;
import javafx.scene.control.Tooltip;
import javafx.scene.image.ImageView;
import javafx.scene.input.MouseEvent;
import org.opencv.core.Core;
import Connectors.*;
import Helpers.AePlayWave;
import java.io.ByteArrayInputStream;
import java.sql.Connection;
import java.sql.ResultSet;
import java.util.concurrent.Executors;
import java.util.concurrent.ScheduledExecutorService;
import java.util.concurrent.TimeUnit;
import javafx.scene.image.Image;
import javax.swing.JFileChooser;
import javax.swing.JOptionPane;
import javax.swing.filechooser.FileSystemView;
import org.opencv.core.Mat;
import panelScenes.*;
public class CameraController implements EventHandler<MouseEvent>
{
Label camera_play,take_photo,paper_name,paper_code,stretch,start_video;
ImageView main_frame,main_frame_side;
VideoCapture capture;
Tooltip play_kit,tool_stretch;
ScheduledExecutorService timer;
int Increment,check_run;
Connection db=SqliteDb.ConnecrDb();
SceneMethods scene;
boolean cameraActive = false;
public CameraController(Label camera_play,Label take_photo,ImageView
main_frame,Tooltip play_kit,ImageView main_frame_side,Label
paper_name,Label paper_code,Label stretch,Tooltip tool_stretch,Label
start_video){
this.camera_play=camera_play;
this.main_frame=main_frame;
this.take_photo=take_photo;
this.main_frame_side=main_frame_side;
this.play_kit=play_kit;
this.paper_code=paper_code;
this.paper_name=paper_name;
this.stretch=stretch;
this.tool_stretch=tool_stretch;
this.start_video=start_video;
}
public void getCameraDetails(){

//............................initialising
classes..............................
System.loadLibrary(Core.NATIVE_LIBRARY_NAME);
capture = new VideoCapture();
scene=new SceneMethods();
//............................setting
listeners...................................
camera_play.setOnMouseClicked(this);
take_photo.setOnMouseClicked(this);
stretch.setOnMouseClicked(this);
start_video.setOnMouseClicked(this);
//..................setting values in the image....
main_frame.setFitWidth(Double.max(450,450));
main_frame.setFitHeight(Double.max(450,450));
}
@Override
public void handle(MouseEvent event) {

[... 218 lines cut ...]

// if the frame is not empty, process it
if (!frame.empty())
{
// convert the image to gray scale
//AnyTime can be used...Imgproc.cvtColor(frame, frame,
Imgproc.COLOR_BGR2GRAY);
// convert the Mat object (OpenCV) to Image (JavaFX)
imageToShow = mat2Image(frame);
}}
catch (Exception e)
{
// log the error
System.err.println("Exception during the image elaboration: " + e);
}}
return imageToShow;
}

private Image mat2Image(Mat frame)
{
MatOfByte buffer = new MatOfByte();
Imgcodecs.imencode(".png", frame, buffer);
return new Image(new ByteArrayInputStream(buffer.toArray()));
}
}

vascubrian, 2016-04-25:
*Im using opencv 3.1 i take photos but i ve failed to use it for recording
video*

vascubrian, 2016-04-25:
If You can help me ,I will be very happy.

sarxos, 2016-04-25:
Hi @vascubrian,

This example demonstrates how to record video using Xuggler (with *no audio*):

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-video-recording/src/main/java/com/github/sarxos/webcam/Encoder.java

To add audio you have to find out how to access audio samples in Xuggler and use encodeAudio(..) in the same way as you use encodeVideo(..) method. I would help, but I have no practical expertise in this area (I never worked with audio in Java). Also, I see you are using OpenCV which I know just briefly.

vascubrian, 2016-04-25:
May you accept mai invite sent u on hangout, and we chat ok

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-04-25:
How can i use  dat example u ve given to mi in the javafx on button click,
when the camera is started.

sarxos, 2016-04-25:
After you read the mentioned example, you should have an idea how to implement similar mechanism in your own code:

1. Get image from camera (using OpenCV, GStreamer, Webcam Capture, or whatever is easier),
2. Convert it to 3-byte BGR ```BufferedImage```,
3. Convert the resultant image to ```IVideoPicture``` frame, and finally
4. Pass resultant frame to ```IMediaWriter``` to method ```encodeVideo(0, frame)```.

I suggest you to check Xuggler or OpenCV forums because this project (Webcam Capture API, where you opened the ticket) is not about video recording.

closed 2016-04-25 by sarxos

vascubrian, 2016-04-26:
Any link for what you have mentioned

vascubrian, 2016-04-26:
hello sarox , if u can help mi and get complete code for recording video using Opencv 3.1 

or start recording audio and video at same xuggler,then combine them after

sarxos, 2016-04-26:
Hi @vascubrian, 

You should ask for help in OpenCV or Xuggler forums. In regards to link - there are many examples to be [found on Google](https://www.google.pl/?client=ubuntu#channel=fs&q=xuggler+record+video+and+audio&gfe_rd=cr). You can also use examples provided by Xuggler:

https://github.com/artclarke/xuggle-xuggler/tree/master/src/com/xuggle/mediatool/demos

Take care!

vascubrian, 2016-05-13:
when i start recording video ,camera view in the panel stops also web cam
java

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-05-13:
writer.addVideoStream(0, 0,
ICodec.ID.CODEC_ID_H264,webcam.getViewSize().width,webcam.getViewSize().height);
long start = System.currentTimeMillis();
for (int i = 0; i <webcam.getFPS(); i++) {
System.out.println("Capture frame " + i);
BufferedImage image = ConverterFactory.convertToType(webcam.getImage(),
BufferedImage.TYPE_3BYTE_BGR);
IConverter converter = ConverterFactory.createConverter(image,
IPixelFormat.Type.YUV420P);
IVideoPicture frame = converter.toPicture(image,
(System.currentTimeMillis() - start) * 1000);
frame.setKeyFrame(i == 0);
frame.setQuality(0);
writer.encodeVideo(0, frame);
}

but also i ve tried threed.sleep(100)...

when start recording camera display in the panel stops,untill video finishes

please help mi

sarxos, 2016-05-14:
Try ```webcam.open(true)``` to start webcam in non-blocking mode.

vascubrian, 2016-05-14:
Webcam b=Webcam.open(true);

gives error in

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-05-14:
1030 [JavaFX Application Thread] INFO com.github.sarxos.webcam.Webcam -
WebcamDefaultDriver capture driver will be used
1295 [atomic-processor-1] INFO
com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam HP Webcam 0
com.github.sarxos.webcam.WebcamException: Cannot execute task
    at
com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:62)
    at
com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:130)
    at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:38)
    at
com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
    at com.github.sarxos.webcam.Webcam.open(Webcam.java:272)
    at startCamera.WriteVideoFile.getVideoFile(WriteVideoFile.java:23)
    at startCamera.CameraController.handle(CameraController.java:82)
    at startCamera.CameraController.handle(CameraController.java:28)
    at
com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
    at
com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
    at
com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
    at
com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
    at
com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
    at
com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
    at
com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
    at
com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
    at
com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
    at
com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
    at
com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
    at
com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
    at
com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
    at
com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
    at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
    at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
    at javafx.event.Event.fireEvent(Event.java:198)
    at javafx.scene.Scene$ClickGenerator.postProcess(Scene.java:3470)
    at javafx.scene.Scene$ClickGenerator.access$8100(Scene.java:3398)
    at javafx.scene.Scene$MouseHandler.process(Scene.java:3766)
    at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
    at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
    at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
    at
com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
    at
com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
    at java.security.AccessController.doPrivileged(Native Method)
    at
com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
    at
com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
    at
com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
    at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
    at com.sun.glass.ui.View.notifyMouse(View.java:937)
    at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
    at
com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
    at java.lang.Thread.run(Thread.java:745)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native
grabber!
    at
com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:264)
    at
com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
    at
com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:71)
    at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
    at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
    ... 1 more

vascubrian, 2016-05-14:
i want keep on displaying camera on the panel and keep on recording but i
ve failed

Any Help please.

Camera part is the one remaing in the my project

displaying camera as im recording any example

sarxos, 2016-05-14:
Your code is wrong:

```
Webcam b=Webcam.open(true);
```

This one is correct:

```
Webcam webcam = Webcam.getDefault();
webcam.open(true);
```

vascubrian, 2016-05-15:
displaying camera as im recording any example

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-05-16:
import org.bytedeco.javacpp.avutil;
import org.bytedeco.javacpp.opencv_core.IplImage;
import org.bytedeco.javacv.CanvasFrame;
import org.bytedeco.javacv.FFmpegFrameRecorder;
import org.bytedeco.javacv.OpenCVFrameGrabber;
public class PanelFrame extends javax.swing.JFrame {
    /** Creates new form PanelFrame */
    public PanelFrame() {
        initComponents();
        getVideo();
    }
public void getVideo(){
       try{
        OpenCVFrameGrabber grabber = new OpenCVFrameGrabber(0);
        grabber.start();
        IplImage grabbedImage = grabber.grab();
        CanvasFrame canvasFrame = new
CanvasFrame("Recording--------------Close to store and Upload File");

canvasFrame.setCanvasSize(grabbedImage.width(),grabbedImage.height());

        //.....................adding it to panel.........................
        jPanel1.add(canvasFrame);
        canvasFrame.validate();
        canvasFrame.repaint();

//.......................brings ERROR.............................

        System.out.println("framerate = " + grabber.getFrameRate());
        grabber.setFrameRate(grabber.getFrameRate());
        FFmpegFrameRecorder recorder = new
FFmpegFrameRecorder("brian.mp4",
grabber.getImageWidth(),grabber.getImageHeight());

        recorder.setVideoCodec(13);
        recorder.setFormat("mp4");

        recorder.setFrameRate(30);
        recorder.setVideoBitrate(10 * 1024 * 1024);

        recorder.start();
        while (canvasFrame.isVisible() && (grabbedImage = grabber.grab())
!= null) {
        canvasFrame.showImage(grabbedImage);
        recorder.record(grabbedImage);
        }
        recorder.stop();
        grabber.stop();
        canvasFrame.dispose();
       }catch(Exception out){
   out.printStackTrace();
       }
}

    @SuppressWarnings("unchecked")
    // <editor-fold defaultstate="collapsed" desc="Generated Code">
    private void initComponents() {

        jPanel1 = new javax.swing.JPanel();
        jButton1 = new javax.swing.JButton();

        setDefaultCloseOperation(javax.swing.WindowConstants.EXIT_ON_CLOSE);

        javax.swing.GroupLayout jPanel1Layout = new
javax.swing.GroupLayout(jPanel1);
        jPanel1.setLayout(jPanel1Layout);
        jPanel1Layout.setHorizontalGroup(

jPanel1Layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGap(0, 430, Short.MAX_VALUE)
[... 84 lines cut ...]
        //</editor-fold>

        /* Create and display the form */
        java.awt.EventQueue.invokeLater(new Runnable() {

            public void run() {
                new PanelFrame().setVisible(true);
            }
        });
    }
    // Variables declaration - do not modify
    private javax.swing.JButton jButton1;
    private javax.swing.JPanel jPanel1;
    // End of variables declaration
}

        recorder.start();
        while (canvasFrame.isVisible() && (grabbedImage = grabber.grab())
!= null) {
            canvasFrame.showImage(grabbedImage);
            recorder.record(grabbedImage);
        }
        recorder.stop();
        grabber.stop();
        canvasFrame.dispose();

sarxos, 2016-05-16:
Hello, 

I thought that you are using Webcam Capture API, but in your code I see that you are using OpenCV binding ([JavaCV](https://github.com/bytedeco/javacv) by [bytedeco](https://github.com/bytedeco)) to capture image from camera. This is ok, however if you have problems with JavaCV you should open a ticket JavaCV project, not in Webcam Capture API project.

https://github.com/bytedeco/javacv

I'm not really familiar with JavaCV and cannot give you any hints explaining why your code is not working as expected.

vascubrian, 2016-05-16:
i ve failed to use webcam api cause
I canot record video while seeing preview in the JPanel at same time

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

vascubrian, 2016-05-17:
*       //.................................RECORDING VIDEO FOR 20 SECONDS
AND CLOSE  canvasFrame----------------*

package startCamera;
import org.bytedeco.javacpp.opencv_core.IplImage;
import static org.bytedeco.javacpp.opencv_highgui.waitKey;
import org.bytedeco.javacv.CanvasFrame;
import org.bytedeco.javacv.FFmpegFrameRecorder;
import org.bytedeco.javacv.OpenCVFrameGrabber;
public class CameraTest {
OpenCVFrameGrabber grabber;
CanvasFrame canvasFrame;
public void getVideo(){
    try{
         grabber = new OpenCVFrameGrabber(0);
        grabber.start();
        IplImage grabbedImage = grabber.grab();
        canvasFrame = new CanvasFrame("Recording--------------Close to
store and Upload File");

canvasFrame.setCanvasSize(grabbedImage.width(),grabbedImage.height());
        System.out.println("framerate = " + grabber.getFrameRate());
        grabber.setFrameRate(grabber.getFrameRate());
        FFmpegFrameRecorder recorder = new
FFmpegFrameRecorder("brian.mp4",
grabber.getImageWidth(),grabber.getImageHeight());
        recorder.setVideoCodec(13);
        recorder.setFormat("mp4");
        recorder.setFrameRate(30);
        recorder.setVideoBitrate(10 * 1024 * 1024);
        recorder.start();
       //.................................RECORDING VIDEO FOR 20 SECONDS
AND CLOSE  canvasFrame----------------

        while (canvasFrame.isVisible() && (grabbedImage = grabber.grab())
!= null) {
        canvasFrame.showImage(grabbedImage);
        recorder.record(grabbedImage);
        }

        recorder.stop();
        grabber.stop();
        canvasFrame.dispose();
       }catch(Exception out){
   out.printStackTrace();
       }
}
}